balena-os/balena-raspberrypi #1: Splash

agherzan, 2016-02-12:
# Steps to reproduce issue 
Connect to the HDMI connected to a monitor
Boot the device

# Expected result 
Resin logo splash screen should be visible

# Actual result 
RaspberryPi: No splash screen at boot

# Other information 
Reported by: Andrei Gherzan 

Test configuration: RaspberryPi 

Test case number: TC07 

Test run: [https://resinio.testlodge.com/projects/16238/runs/192879?tab=2&run_section_id=168778&executed_case_id=9333888#executed_case_9333888](https://resinio.testlodge.com/projects/16238/runs/192879?tab=2&run_section_id=168778&executed_case_id=9333888#executed_case_9333888)

telphan, 2016-03-04:
Is this issue present on Raspberrypi or Raspberrypi2?

agherzan, 2016-03-04:
Rpi1

telphan, 2016-03-04:
Well, this is not exactly a bug. The reason for the splash not showing on rpi1 is slow hardware. It takes too long to show the splash. Therefore the boot finishes before plymouth has a chance of showing splash.

agherzan, 2016-03-04:
Yes. We are aware of this but maybe we could optimize boot to inject the splash service sooner in the boot seq services?

telphan, 2016-03-04:
There is only one way that springs to mind, but that involves making use of an initramfs.

lurch, 2016-10-12:
The [new version of Raspbian](https://www.raspberrypi.org/blog/introducing-pixel/) includes a splashscreen, maybe there's code you can borrow from there?

agherzan, 2016-10-12:
Sounds interesting. We will take a look on this.

closed 2016-10-17 by agherzan