balena-os/balena-raspberrypi #120: update kernel to 4.9

shaunmulligan, 2017-08-28:


floion, 2017-08-28:
I think this should be done from the bsp instead of doing it here

agherzan, 2017-09-04:
I agree.

willnewton, 2017-09-06:
It looks like Pyro uses 4.9 by default.

agherzan, 2017-09-06:
@willnewton the request is actually for 4.9.29+.

agherzan, 2017-09-08:
`meta-raspberrypi` pyro comes with 4.9.41 which is the latest version currently available. @willnewton can you link this to the pyro update task so we can close it as soon as that is done?

willnewton, 2017-09-08:
This should be fixed when #126 is closed

willnewton, 2017-10-27:
The Pyro PR has been merged so this is completed.

closed 2017-10-27 by willnewton

qerub, 2017-10-30:
About when can one expect a Resin OS release with this kernel?

qerub, 2017-10-30:
"@qerub unassigned @willnewton 37 seconds ago"

Uhm… Did I do something stupid or just trigger a GitHub bug?

floion, 2017-10-31:
No idea what happened with github above.
Anyway, there should be a release for rpi3 in the following days. Stay tuned