balena-os/balena-raspberrypi #131: recipe jq-1.5-r0 : failed to compile

FXRer, 2017-10-02:
Hi all,

I tried to make an image for raspberry pi 3 target under master branch (by using _./resin-yocto-scripts/build/barys -r --shared-downloads $(pwd)/shared-downloads/ --shared-sstate $(pwd)/shared-sstate/ -m raspberrypi3_ command)  and I experienced the following problem:
...
NOTE: Running task 1755 of 3227 (~/Workspace/GIT/resinos/resin-raspberrypi/build/../layers/meta-openembedded/meta-oe/recipes-devtools/jq/jq_1.5.bb:do_compile)
NOTE: recipe jq-1.5-r0: task do_compile: Started
ERROR: jq-1.5-r0 do_compile: oe_runmake failed
ERROR: jq-1.5-r0 do_compile: Function failed: do_compile (log file is located at ~/Workspace/GIT/resinos/resin-raspberrypi/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/jq/1.5-r0/temp/log.do_compile.31306)
ERROR: Logfile of failure stored in: /home/khang/Workspace/GIT/resinos/resin-raspberrypi/build/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/jq/1.5-r0/temp/log.do_compile.31306
NOTE: recipe jq-1.5-r0: task do_compile: Failed
ERROR: Task (/home/khang/Workspace/GIT/resinos/resin-raspberrypi/build/../layers/meta-openembedded/meta-oe/recipes-devtools/jq/jq_1.5.bb:do_compile) failed with exit code '1'
...

Could anyone have a solution, please?

Best regards,
Pascal
[20171002045420.log](https://github.com/resin-os/resin-raspberrypi/files/1348761/20171002045420.log)
[console-latest.log](https://github.com/resin-os/resin-raspberrypi/files/1348762/console-latest.log)

agherzan, 2017-10-02:
Hi @FXRer ,

Can you attach the compile log as it is referenced in the log?

Thanks,
Andrei

FXRer, 2017-10-02:
Hi Andrei,
I already the log files at the end of my previous comment. 
Best regards,
Pascal

agherzan, 2017-10-02:
I downloaded both of your file but I can't see anything that would explain the error. Did I miss something?

floion, 2017-10-03:
Hi Pascal, can you paste here the command you used to clone the resin-raspberrypi repository?

Also, can you try and build using resin-yocto-scripts/automation/jenkins_build.sh script? Run this script with -h and see the required usage. This script builds in a docker container so you should be able to build it this way if there's a problem building it directly on your development machine

FXRer, 2017-10-03:
HI Florion,

The command I used to clone the **resin-raspberrypi** repository is :
_$ git clone https://github.com/resin-os/resin-raspberrypi_

For your suggestion of using resin-yocto-scripts/automation/jenkins_build.sh script, I will give it a try and update the result ASAP.
Best regards,
Pascal

floion, 2017-10-03:
Hi Pascal,
the correct clone command should be:

git clone --recursive https://github.com/resin-os/resin-raspberrypi

Can you try to re-clone using the above command and build it again using barys?

FXRer, 2017-10-04:
Hi Florion,
Thanks fr your help. your solution works well.

closed 2017-10-04 by agherzan