balena-os/balena-raspberrypi #144: Could you note in the change log when an update includes a linux kernel version change?

BrianAdams, 2017-11-09:
I have a custom kernel module that needs to get recompiled and it would help to be able to scan the notes to know when the kernel is updated (which triggers the need to recompile).

agherzan, 2017-11-09:
Hi @BrianAdams . Usually the update comes from the BSP layer update. It's hard to trace the changelog of each of the layer dependencies we have. So generally any update of the meta-raspberrypi layer could come with an updated kernel version. I say generally because we can decide to update it due to issue in the meta-resin-raspberrypi temporary until the BSP catches up. I will raise this internally though because the request sounds valid to me.

floion, 2017-11-10:
How are you adding the kernel module? Do you compile it before hand and just copy it in your container?

floion, 2017-11-10:
I guess if you could do it like this https://github.com/resin-io-playground/kernel-module-build it would be a matter of re-pushing your app to the device

agherzan, 2017-11-27:
We discussed this internally and in the future we will be providing package versions along with the OS images.

closed 2019-02-28 by floion