balena-os/balena-raspberrypi #153: Pi with dev OS will not boot with Hyperpixel overlays enabled

pimterry, 2017-12-22:
Discussion: https://www.flowdock.com/app/rulemotion/resin-projects/threads/OQ1PVEwq2mpdX4HOvH4qI93Da0M

The [hyperpixel](https://github.com/pimoroni/hyperpixel) should be supported, and does work with a prod OS for both the pi 3 & pi zero, if you add the overlays and config.txt lines from that readme. However, if you follow the same process on a dev build then the OS does not boot, and after a brief initial blinking the ACT light goes solid green.

This seems to be caused by the `enable_uart=1` line in config.txt. If that is removed, dev builds then boot.

Unclear whether this is an actual bug, or simply something that needs to be better documented.

drewcovi, 2018-08-06:
@pimterry how did you get the hyperpixel installed on your instance?

pimterry, 2018-08-08:
No, I didn't, in the end for my project I just moved to a Pi 3. Good reminder though! 

I've just given this a quick test, and my pi zero now seems to boot fine with a hyperpixel attached (using a dev build of 2.12.7). The screen powers up fine, but I haven't got a full project & config available to test that the output actually works correctly. @drewcovi do you see this working for you, or are you still having issues?

drewcovi, 2018-08-14:
@pimterry i cant even get it running on a pi 3... 😢 

Resin OS 2.13.6+rev1

```
Connecting to bc5081abf54b5eb6e6cf9bdeda1f00ea...
Spawning shell...
=============================================================
    Welcome to ResinOS
=============================================================
root@bc5081a:~# cat /mnt/boot/config.txt
display_default_lcd=1
display_rotate=0
dpi_group=2
dpi_mode=87
dpi_output_format=0x6f016
dtoverlay=hyperpixel
enable_dpi_lcd=1
framebuffer_height=480
framebuffer_width=800
hdmi_timings=800 0 50 20 50 480 1 3 2 3 0 0 0 60 0 32000000 6
overscan_bottom=0
overscan_left=0
overscan_right=0
overscan_top=0
disable_splash=1
dtparam=i2c_arm=off
dtparam=spi=off
dtparam=audio=on
gpu_mem_1024=512
root@bc5081a:~# ls -latr /mnt/boot/overlays/hyperpixel*
-rwxr-xr-x 1 root root 1646 Aug  6 11:05 /mnt/boot/overlays/hyperpixel.dtbo
-rwxr-xr-x 1 root root  612 Aug  6 11:05 /mnt/boot/overlays/hyperpixel-gpio-backlight.dtbo
root@bc5081a:~#
```

drewcovi, 2018-08-15:
@pimterry to be clear: you have power but no actual output?

That sounds like what I’m (not) seeing.

Sure. It runs. But the screen is dark.

drewcovi, 2018-08-16:
@pimterry making progress. fyi: had to bundle the hyperpixel-init script into my image. i believe it only has to be run one time. but after that, I'm up and running.

pimterry, 2018-08-17:
Great! Glad you made progress. Sorry I haven't had much time to test this out myself. Are you saying it works 100% though, on a dev image, so we can close this issue?

floion, 2019-10-23:
Closing because no answer was received for some time

closed 2019-10-23 by floion

suddenelfilio, 2021-08-04:
is there documentation on how to get this configured? not that familiar with all this :-)