balena-os/balena-raspberrypi #154: Add sd8887_uapsta.bin wifi firmware to the raspberrypi OS image

floion, 2017-12-23:
Check if the current	sd8887_uapsta.bin firmware in the pi3 image is the latest. If it is not, then bump the linux-firmware revision to latest available (through an append similar to https://github.com/resin-os/resin-intel/blob/master/layers/meta-resin-genericx86/recipes-kernel/linux-firmware/linux-firmware_git.bbappend where you only specify the new revision) and then add the corresponding package here https://github.com/resin-os/resin-raspberrypi/blob/master/layers/meta-resin-raspberrypi/recipes-core/packagegroups/packagegroup-resin-connectivity.bbappend to make sure this firmware makes its way into the rootfs

spanceac, 2017-12-29:
I created this PR: https://github.com/resin-os/resin-raspberrypi/pull/155 which adds the sd8887_uapsta.bin firmware in the final image.
This isn't the latest wi-fi firmware that exists in the linux-firmware git.
A newer version exists, added by the commit "5e523f5cd672d8fae7cdbb147d0b1fb31119414d".
The linux-firmware package from poky, is at an older SRCREV.
If we want to get the latest firmware, then we need to create a new package just for this firmware or update the whole linux-firmware recipe.

floion, 2017-12-29:
@spanceac in case the firmware is not the newest, you can add the desired SRCREV in a .bbappend. See my comment above. Have you tried that and it did not work?

spanceac, 2017-12-29:
@floion I tried to update the SRCREV of linux-firmware in a bbappend, but the linux-firmware package build stared to throw errors. That's why I suggested that the whole recipe would need to get an update or we could create a recipe only for the sd8887_uapsta.bin firmware.

floion, 2018-01-05:
Fixed by https://github.com/resin-os/resin-raspberrypi/pull/155

closed 2018-01-05 by floion