balena-os/balena-raspberrypi #157: enable_uart=1 does not make bt work when using ttyS0

floion, 2018-01-08:
Switch bt to use ttyS0 (by using "dtoverlay=pi3-miniuart-bt" in config txt) and then also use "enable_uart=1"  in config.txt and bt will not work. If instead one uses "force_turbo=1" or "core_g=freq=250" in config.txt then bt works on ttyS0.

Needs investigation on why "enable_uart=1" it not enough to make bt work (https://www.raspberrypi.org/documentation/configuration/uart.md)

agherzan, 2018-01-08:
This is expected. See: https://www.raspberrypi.org/forums/viewtopic.php?f=107&t=138223&start=75#p1041713 .

floion, 2018-01-08:
So closing this item as per my initial conclusions before opening this issue.

closed 2018-01-08 by floion

agherzan, 2018-01-08:
At that point we didn't have the explaination. And there is still a question about the baudrate.