balena-os/balena-raspberrypi #168: Unifying all devices onto using resin-rootA/rootB for root partition

agherzan, 2018-02-06:
_From @imrehg on December 1, 2017 22:54_

Not all the boards use labels when selecting the root partition. For example raspberrypi uses root=mmcblk0pX .

_Copied from original issue: resin-os/meta-resin#918_

agherzan, 2018-02-06:
@imrehg  meta-resin provides support for LABEL based root kernel arguments. See: https://github.com/resin-os/meta-resin/blob/master/meta-resin-common/recipes-core/initrdscripts/files/rootfs#L34 . It means that this issue needs to be moved to the raspberrypi repository as changes to cmdline.txt are needed. Could you please create an issue in resin-raspberrypi and close this one?

floion, 2018-04-11:
Assigned Theodor as per a discussion on 6th of March here: https://github.com/resin-os/resin-raspberrypi/pull/173

floion, 2018-04-11:
@telphan , I have recreated to PR here: https://github.com/resin-os/resin-raspberrypi/pull/189
Please review and test it on the board

floion, 2018-04-27:
Fixed by https://github.com/resin-os/resin-raspberrypi/pull/189

closed 2018-04-27 by floion