balena-os/balena-raspberrypi #178: RPI3 bluetooth fails at start up (occasionally)

shaunmulligan, 2018-03-09:
I shut the device down and the power cycled it and the Bluetooth interface never initialised correctly. These are the logs from dmesg:
```
root@a6cfd31:~# dmesg | grep Blue*
[   15.275877] Bluetooth: Core ver 2.22
[   15.291339] Bluetooth: HCI device and connection manager initialized
[   15.291473] Bluetooth: HCI socket layer initialized
[   15.291482] Bluetooth: L2CAP socket layer initialized
[   15.291539] Bluetooth: SCO socket layer initialized
[   15.297374] Bluetooth: HCI UART driver ver 2.3
[   15.297378] Bluetooth: HCI UART protocol H4 registered
[   15.297380] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   15.297492] Bluetooth: HCI UART protocol Broadcom registered
[   17.368161] Bluetooth: hci0 command 0x0c14 tx timeout
[   19.449713] Bluetooth: hci0 command 0x0c25 tx timeout
[   19.866079] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   19.871695] Bluetooth: BNEP filters: protocol multicast
[   19.888387] Bluetooth: BNEP socket layer initialized
[   21.528158] Bluetooth: hci0 command 0x0c38 tx timeout
[   23.608167] Bluetooth: hci0 command 0x0c39 tx timeout
[   25.688217] Bluetooth: hci0 command 0x1003 tx timeout
[   27.768211] Bluetooth: hci0 command 0x1001 tx timeout
[   29.848213] Bluetooth: hci0 command 0x1009 tx timeout
[   37.848157] Bluetooth: hci0 command 0x1003 tx timeout
[   39.928160] Bluetooth: hci0 command 0x1001 tx timeout
[   42.008162] Bluetooth: hci0 command 0x1009 tx timeout
```

It seems related to https://github.com/agherzan/meta-raspberrypi/issues/148

floion, 2018-07-30:
@shaunmulligan can you retest this please and update this ticket?

shaunmulligan, 2018-08-09:
@floion I haven't been able to reproduce this again :(

floion, 2018-11-07:
@shaunmulligan can this be closed now since you cannot reproduce it on latest versions?

shaunmulligan, 2018-11-08:
@floion yup I'll close it, thanks

closed 2018-11-08 by shaunmulligan