balena-os/balena-raspberrypi #181: Why is armv7 and not armv8/aarch64?

lasote, 2018-03-14:
Why is that?

Raspberry Pi 3 | raspberrypi3 | armv7hf | -
-- | -- | -- | --

floion, 2018-03-14:
At this moment we are not building a 64 bit OS for the PI3
Are you missing anything in particular because of this?

lasote, 2018-03-14:
Not really, just curious. So the OS is not 64 bit. Understood.

lasote, 2018-03-15:
Hi again, why the `resin/armv7hf-python` returns this to the `uname -a`? 

Linux ea712bd312c0 4.12.13 #1 SMP Thu Sep 14 13:00:05 UTC 2017 **armv8l** GNU/Linux

guillaumekh, 2018-06-14:
>At this moment we are not building a 64 bit OS for the PI3
>Are you missing anything in particular because of this?

You got a prospective customer here who does ✋ 
We need to run MongoDB 3.x which is 64bit-only.

floion, 2018-06-15:
Hi,
we'll revisit this and come back to you

pesicsasa, 2018-06-15:
You got a prospective customer here who does x2 (:
We need to use Hyperlledger Fabric, and Hyperledger Fabric only supports 64-bit architectures. Would be a charm to use Resin for this. 
Thanks!

pcarranzav, 2018-06-19:
Hi @guillaumekh @pesicsasa would you contact us at solutions at resin.io to chat about this? Thanks!

pesicsasa, 2018-06-26:
@pcarranzav please check your inbox at solutions@resin.io

mcfilib, 2018-08-02:
@pcarranzav I'd also be interested in a 64bit ResinOS.

benharosh, 2018-10-05:
Hi we're also facing some issures trying to run mongodb on ResinOS. MongoDB supports only 64-bit arch, while ResinOS is running the armv7l (32-bit).

fignew, 2018-11-23:
Fedora no longer supports arm32. Only aarch64. One can no longer run Fedora containers on Balena.

wlisac, 2018-12-19:
Also interested in 64bit balenaOS. I'd like to be able to run a 64bit version of Swift lang.

floion, 2019-02-28:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/297

wlisac, 2019-03-05:
@floion is there a docker hub image we can start using for this? I'm currently using `balenalib/raspberrypi3-debian:stretch` for 32 bit.

floion, 2019-03-06:
@nghiant2710 have the base images been deployed?

nghiant2710, 2019-03-06:
@floion, they're ready, just change the device-type to `raspberrypi3-64` to use the armv8/aarch64 base images

floion, 2019-03-06:
Thanks Trong. @wlisac ^

steevegldev, 2019-03-09:
This information is still not available in https://www.balena.io/docs/reference/base-images/base-images-ref/ ;)

OdysLam, 2019-03-15:
Hey,  I am trying to use  balenalib/raspberrypi3-64-ubuntu but every RUN command will produce a "standard_init_linux.go:195: exec user process caused "exec format error"

Guess it's not yet functional?

nghiant2710, 2019-03-15:
@steevegldev, it's a bit outdated, we will update the reference soon. Thanks.

@OdysLam, where did you run that docker image? it's an `aarch64` image so if you run it on a x64/x86-64 device then you will get `exec format error`.

OdysLam, 2019-03-15:
@nghiant2710 I got the error when pushing the dockerfile on a Rpi3 in my local network.

Results from local scan: 

```Reporting scan results
- 
  host:          //////
  address:       192.168.1.2
  dockerInfo: 
    Containers:        6
    ContainersRunning: 1
    ContainersPaused:  0
    ContainersStopped: 5
    Images:            16
    Driver:            aufs
    SystemTime:        2019-03-15T09:12:21.148557905Z
    KernelVersion:     4.14.79
    OperatingSystem:   balenaOS 2.29.2+rev2
    Architecture:      armv7l
  dockerVersion: 
    Version:    17.12.0-dev
    ApiVersion: 1.35
```

OdysLam, 2019-03-19:
Hey @nghiant2710, if I understand correctly, using the base image as 64 bits is not the same as running balenaOS in 64 bits. How do I run balenaOS in 64 mode?

Because i have some already built containers which will run only on 64 ARM arch, so I suppose I need balenaOS in 64.

nghiant2710, 2019-03-19:
@floion, can you please point @OdysLam to the raspberrypi3-64 balenaOS image? I'm not sure the images we provide here https://www.balena.io/os/ is armv7 or aarch64.

floion, 2019-03-21:
@OdysLam we have a 64 bits OS image for Raspberry Pi3. It is under testing currently, it should be released shortly. Please stay tuned, will update this ticket once it's available.

OdysLam, 2019-03-21:
@floion thank you for your swift response. Really looking forward with this!

wlisac, 2019-03-24:
@floion Thank you.

Just to make sure I understand this correctly: 64 bit base images are available, but it's not possible to run them with Balena until balenaOS is available in 64 bit, which is coming soon. Does that sound right? 🙂

sgammon, 2019-03-31:
@floion any update on this? i need to run 64 bit containers, but i'm getting the error @OdysLam is getting - about exec format errors.

OdysLam, 2019-03-31:
@sgammon Apparently, the error has nothing to do with the fact that BalenaOS is 32 bits. I managed to build my containers using the --emulate command in balena-cli and loaded them on the device. Balena says that the services and containers were up and running but the services didn't seem to work.

As far as I could tell, the error had to do with the fact that I was building the images on my mac for another architecture without emulating the needed architecture. BalenaOS 32 means that the containers will be installed and all, but the services inside them will produce errors because they need 64 environment (more ram, etc.). 

At least this is what I got to and I stopped looking into it until we got some news.

Cheers!

sgammon, 2019-03-31:
Thank you for the quick reply @OdysLam 

I saw your comments but I'm building via the `balena push <appname>` command, up to the cloud using Balena's CI services. I need 64-bit support in the container for Java gRPC which uses `tcnative` or `conscrypt`, and both only distribute 64-bit binaries.

sgammon, 2019-03-31:
I'm curious about how to get this to work @floion @nghiant2710, or if this might be my fault in how I'm building it or what not. Thank you in advance, we're really excited about Balena on Pi

sgammon, 2019-03-31:
Sorry @OdysLam, just to clarify, were you able to get your 64bit containers to work that way, or no? Lol

OdysLam, 2019-03-31:
Yes, I was able to build containers using the ubuntu64 for Rpi base image and ship them using the local dev mode of balena-cli. I used the --emulate command to emulate the architecture and build them on my laptop. Balena reports that containers are up and running, but I think the services inside them don't, as they expect a 64 env.

sgammon, 2019-04-01:
i can't get it to work with any 64bit Pi image, be it `alpine`, `ubuntu`, `debian`, `openjdk` or other variants, and so on. going to keep trying

also tried the `emulate` flag and pushing locally but, no dice.

thank you again for your help

steevegldev, 2019-04-01:
64bits container require 64bits OS...

floion already answer. The image is not released yet. You have to build the yocto image yourself. But it's not easy, you have to know yocto has well or be patient.

floion, 2019-04-01:
Hi all, we should be ready to deploy the 64 bits OS image for the pi3 today or tomorrow if nothing goes wrong.
Once that is done, if you are using templated Dockerfiles then you will also be using the 64 bits base images as @nghiant2710 said above.
I will let you know here when this is in production

OdysLam, 2019-04-01:
@floion these are amazing news! Can't wait to try these.

Any idea why I was able to build ubuntu64 images, use already-built images that I know for sure work in 64 env but I was able to ship them to Rpi3 and balena informed me that they worked alright?

shaunmulligan, 2019-04-01:
@OdysLam if you pushed them through the build server, those would be be buildable on our servers because they are capable of building aarch64 as the CPU is actually an aarch64 CPU under the covers, then the supervisor would be able to download and the containers, but won't be able to run them because all of the binaries would require 64 bit CPU instruction set, which the armv7 (32 bit) variant of the RPI3 (our default) would not be able to run. 

As florin said, this week we will be releasing a new device type called raspberrypi3-64, which will have the hostOS and kernel compiled to run 64 bit.

sgammon, 2019-04-02:
thank you @floion, @shaunmulligan, and everyone else for your help, super excited!

floion, 2019-04-03:
Hi everybody, the 64 bit variant for rpi3 is now available.
Please give it a spin

OdysLam, 2019-04-03:
> Hi everybody, the 64 bit variant for rpi3 is now available.
> Please give it a spin

Excuse me for my cluelessness, but in order to run 64bit images (i.g Edgex) we need BalenaOS to be 64, am I right? thus I need to re-download BalenaOS.

marvinroger, 2019-04-03:
@OdysLam 

> the supervisor would be able to download and the containers, but won't be able to run them because **all of the binaries would require 64 bit CPU instruction set, which the armv7 (32 bit) variant of the RPI3 (our default) would not be able to run**.

So yes, in order to run 64 bit images (without QEMU), you need the OS to be 64 bit.

steevegldev, 2019-04-03:
> Hi everybody, the 64 bit variant for rpi3 is now available.
> Please give it a spin

link?

OdysLam, 2019-04-04:
@marvinroger  Yes I am slowly understanding how docker and balena works. Thank you everyone for this amazing work.
Will make sure to upload, at some time, how Balena can be used to ship EdgeX. Still trying to set it up.

Cheers!

sgammon, 2019-04-04:
Thank you @floion giving it a try today! :)

sgammon, 2019-04-04:
@floion, sorry, where can i find the image? i'm assuming you mean a 64-bit host OS image, but i don't see a 64-bit Pi option on Balena.
[image]

shaunmulligan, 2019-04-04:
@sgammon you will have to create a new application, since the rpi3-64 bit is aarch64, it can't be in an `armv7` app. You can see it listed below:
[image]

sgammon, 2019-04-04:
@shaunmulligan thank you :) that does make sense.

sgammon, 2019-04-04:
@shaunmulligan, @floion: i'm getting issues with qemu errors trying to use the OpenJDK images (for JDK 11, JDK 8 seems to work fine). i filed a bug about it in #325 - thank you again for your work on this

SamurAchzar, 2019-06-25:
Trying to build a Docker image here, but it's always building as 32-bit rather than 64-bit. I'm using this in my Dockerfile:
`FROM balenalib/raspberrypi3-64-debian:latest`

Running balenaOS 2.31.5+rev5
The Host OS shows as aarch64, so do the Docker container when doing uname -a:
`Linux 3c555199a811 4.14.98 #1 SMP PREEMPT Mon Apr 1 15:47:35 UTC 2019 aarch64 GNU/Linux`

But the files in the image are all armhf:
`/lib/arm-linux-gnueabihf/libm-2.24.so: ELF 32-bit LSB shared object, ARM, EABI5`

floion, 2019-06-25:
Are you using the raspberrypi3 64 bits OS image?

SamurAchzar, 2019-06-25:
Yes. 

`
ID="balena-os"
NAME="balenaOS"
VERSION="2.31.5+rev5"
VERSION_ID="2.31.5"
PRETTY_NAME="balenaOS 2.31.5+rev5"
MACHINE="raspberrypi3-64"
VARIANT="Development"
VARIANT_ID="dev"
RESIN_BOARD_REV="97f1273"
META_RESIN_REV="64859b0"
SLUG="raspberrypi3-64"
`

Page-, 2019-06-25:
@SamurAchzar I just checked balenalib/raspberrypi3-64-debian and get results like
```
# file /lib/aarch64-linux-gnu/ld-2.24.so
/lib/aarch64-linux-gnu/ld-2.24.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[sha1]=a231713f41d92997033e41a5e5e72e01a6f2cca8, stripped
```
and do not have `/lib/arm-linux-gnueabihf/libm-2.24.so` installed by default at all?  How did you install that library so that I can try to reproduce?

SamurAchzar, 2019-06-25:
I did not install that library, it's what I got by default on the base Docker image. 
`FROM balenalib/raspberrypi3-64-debian:latest` is what I have in the Dockerfile.

Page-, 2019-06-25:
This is what I get on the latest image
```
~ $ docker pull balenalib/raspberrypi3-64-debian:latest
latest: Pulling from balenalib/raspberrypi3-64-debian
Digest: sha256:2aa68c0e41adab9bdd774ee1a6d6d8b999cf75cffccb43c8d7746e4b90495bc8
Status: Image is up to date for balenalib/raspberrypi3-64-debian:latest
~ $ docker run -it balenalib/raspberrypi3-64-debian:latest ls /lib -alh
total 40K
drwxr-xr-x 17 root root 4.0K Jun 22 22:57 .
drwxr-xr-x 47 root root 4.0K Jun 25 20:51 ..
drwxr-xr-x  3 root root 4.0K Jun 22 22:57 aarch64-linux-gnu
drwxr-xr-x  2 root root 4.0K Jun 22 22:57 ifupdown
drwxr-xr-x  2 root root 4.0K Jun 10 00:00 init
lrwxrwxrwx  1 root root   28 Feb  6 21:17 ld-linux-aarch64.so.1 -> aarch64-linux-gnu/ld-2.24.so
drwxr-xr-x  3 root root 4.0K Jun 10 00:00 lsb
drwxr-xr-x  2 root root 4.0K Jun 22 22:57 modprobe.d
drwxr-xr-x  6 root root 4.0K Jun 20 16:24 systemd
drwxr-xr-x 15 root root 4.0K Dec 28  2017 terminfo
drwxr-xr-x  6 root root 4.0K Jun 22 22:57 udev
```
Could you try manually pulling the image or using `docker build --pull` to make sure you're using the latest image as it looks like maybe you have an older version cached that was mistakenly built with 32 bit binaries

SamurAchzar, 2019-06-25:
I thought it could be caching too, but the same phenomena appeared both when I built locally and when I built on the balenaCloud servers, different caching I assume?

If you give me your contact info we'll take it offline and perhaps you could look directly on my deployment repo to see what's going on there. I'm fine with 32 bit for now, so no pressure from me, but you may want to investigate further.

Page-, 2019-06-28:
@nghiant2710 is the maintainer for the base images so he's the best to dig into this, could you take over on this please @nghiant2710 ?