balena-os/balena-raspberrypi #206: ERROR: balena-native - gcc: error: unrecognized command line option "-no-pie"

bbinet, 2018-06-20:
I get the following error when trying to build the latest version (2.13.1+rev1) of resin-raspberrypi:
```
ERROR: balena-native-17.06.0-dev+git73136d1985f28a34705f537e90636cf208734048-r0 do_compile: Function failed: do_compile (log file is located at /home/bruno/dev/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.06.0-dev+git73136d1985f28a34705f537e90636cf208734048-r0/temp/log.do_compile.25317)
ERROR: Logfile of failure stored in: /home/bruno/dev/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.06.0-dev+git73136d1985f28a34705f537e90636cf208734048-r0/temp/log.do_compile.25317
Log data follows:
| DEBUG: Executing shell function do_compile
| # WARNING! I don't seem to be running in a Docker container.
| # The result of this command might be an incorrect build, and will not be
| # officially supported.
| #
| # Try this instead: make all
| #
| 
| bundles/17.06.0-dev already exists. Removing.
| 
| ---> Making bundle: dynbinary-balena (in bundles/17.06.0-dev/dynbinary-balena)
| Building: bundles/17.06.0-dev/dynbinary-balena/balena-17.06.0-dev
| Created binary: bundles/17.06.0-dev/dynbinary-balena/balena-17.06.0-dev
| 
| # github.com/docker/docker/cmd/mobynit
| /home/bruno/dev/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.06.0-dev+git73136d1985f28a34705f537e90636cf208734048-r0/recipe-sysroot-native/usr/lib/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
| gcc: error: unrecognized command line option ‘-no-pie’
| 
| WARNING: exit code 2 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/bruno/dev/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.06.0-dev+git73136d1985f28a34705f537e90636cf208734048-r0/temp/log.do_compile.25317)
ERROR: Task (virtual:native:/home/bruno/dev/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-containers/balena/balena_git.bb:do_compile) failed with exit code '1'
```
gcc complains about the `-no-pie` option which it does not support.
Is there any minimum version of gcc required?

bbinet, 2018-06-21:
I think this issue has been introduced by this commit: https://github.com/resin-os/meta-resin/commit/3af7ac7a686d349c1b60953d7ffa8eac14141ce3

bbinet, 2018-06-21:
I've just tried to rebuild my resin image after commenting out the `-no-pie` option at https://github.com/resin-os/meta-resin/blob/master/meta-resin-sumo/recipes-containers/balena/balena_git.bbappend#L3

Then the compilation was successful, but as it was predictable I now face the same runtime issue (`Kernel panic - not syncing: Attempted to kill init!`) that the commit https://github.com/resin-os/meta-resin/commit/3af7ac7a686d349c1b60953d7ffa8eac14141ce3 was aimed to fix.

Any idea how I can fix this `gcc: error: unrecognized command line option ‘-no-pie’` balena compilation issue?

floion, 2018-06-23:
Yes @bbinet you should be updating the gcc on your host machine. I'm not sure what is the gcc version that first added support for "-no-pie". What version do you use?

bbinet, 2018-06-25:
Thanks @floion.
My (old) debian jessie has gcc 4.9, but I thought that yocto was coming with its own compiler.
So I can try to build a newer version of gcc, but do you know if I backport gcc version 6.3.0 from debian stretch would be enough? (I may also take this opportunity to finally move to debian stretch ;-) )

bbinet, 2018-06-25:
Thank you @floion.
I've upgraded to debian stretch (gcc 6), and resin-raspberrypi does compile smoothly now!

closed 2018-06-25 by bbinet

floion, 2018-06-25:
You are welcome, glad it's working now