balena-os/balena-raspberrypi #220: Can't install PostgreSQL on Raspbian Stretch

becklo, 2018-08-06:
Hi, 

I am trying to build a [LoraServer](https://www.loraserver.io/) on a Raspberry Pi 3 with Stretch using resin.io but i cannot install postgresql and I from what I have looked this is due to the fact that this image is build on debian:stretch-slim.

To install it I simply use (I also tried to wget the repository and add the deb file):

    `RUN apt-get update && apt-get install -y --no-install-recommends postgresql (also tried with postgresql-9.6) postgresql-client postgresql-contrib && rm -rf /var/lib/apt/lists/*`

On git push an error is displayed:

    

> [main] Setting up postgresql-client-9.5 (9.5.4-3) …
>     [main] update-alternatives: using /usr/share/postgresql/9.5/man/man1/psql.1.gz to provide /usr/share/man/man1/psql.1.gz (psql.1.gz) in auto mode
>     [main] update-alternatives: error: error creating symbolic link ‘/usr/share/man/man7/ABORT.7.gz.dpkg-tmp’: No such file or directory
>     [main] dpkg: error processing package postgresql-client-9.5 (–configure):
>     [main] subprocess installed post-installation script returned error exit status 2
>     [main] dpkg: dependency problems prevent configuration of postgresql-9.5:
>     [main] postgresql-9.5 depends on postgresql-client-9.5; however:
>     [main] Package postgresql-client-9.5 is not configured yet.
>     [main] dpkg: error processing package postgresql-9.5 (–configure):
>     [main] dependency problems - leaving unconfigured

Which ends up with this message:

  

>   [main] Errors were encountered while processing:
>     [main] postgresql-client-9.5
>     [main] postgresql-9.5
>     [main] postgresql-contrib-9.5
>     [main] E: Sub-process /usr/bin/dpkg returned an error code (1)

And finishes with : **[Error] Not deploying release.**

Did anyone else had this problem ? How can it be fixed ?

Best regards,
becklo

nghiant2710, 2018-08-08:
@floion, @becklo I think the problem is `/usr/share/man/man7` is missing since they do `rm -rf-ed /usr/share/man` when building the image. So just add this line `RUN mkdir -p /usr/share/man/man7` before installing then I think it will work

floion, 2018-08-26:
Thanks @nghiant2710 
@becklo did you try that?

floion, 2019-02-28:
Closing due to lack of response to the proposed solution

closed 2019-02-28 by floion