balena-os/balena-raspberrypi #221: Hyperpixel not working

drewcovi, 2018-08-07:
Any help here would be immensely appreciated... I'm attempting to get the Hyperpixel up and running on the latest ResinOS.

I've seen others have had success, but I'm unable to sort out what to change. I've provided the overlay in the boot folder, but I'm unable to get anything to display when setting display_default_lcd=1

I've attached my custom config variables for reference.

[image]

drewcovi, 2018-08-07:
The resulting config.txt after syncing with resin.io

```
disable_splash=0
display_default_lcd=1
display_rotate=2
dpi_group=2
dpi_mode=87
dpi_output_format=0x6f016
dtparam=audio=on
dtparam=i2c_arm=off
dtparam=spi=off
enable_dpi_lcd=1
framebuffer_height=480
framebuffer_width=800
gpu_mem_1024=512
hdmi_timings=800 0 50 20 50 480 1 3 2 3 0 0 0 60 0 32000000 6
overscan_bottom=0
overscan_left=0
overscan_right=0
overscan_top=0
```

drewcovi, 2018-08-14:
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 489860K/524288K available (8192K kernel code, 598K rwdata, 2264K rodata, 8192K init, 721K bss, 26236K reserved,8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 27776 entries in 82 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000021] Switching to timer-based delay loop, resolution 52ns
[    0.000269] Console: colour dummy device 80x30
[    0.000300] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000318] pid_max: default: 32768 minimum: 301
[    0.000631] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000646] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.001491] CPU: Testing write buffer coherency: ok
[    0.001904] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.002294] Setting up static identity map for 0x100000 - 0x10003c
[    0.002415] Hierarchical SRCU implementation.
[    0.003055] smp: Bringing up secondary CPUs ...
[    0.003757] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.004508] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005244] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.005351] smp: Brought up 1 node, 4 CPUs
[    0.005375] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.005383] CPU: All CPU(s) started in HYP mode.
[    0.005391] CPU: Virtualization extensions available.
[    0.006138] devtmpfs: initialized
[    0.016322] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.016536] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016558] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.017135] pinctrl core: initialized pinctrl subsystem
[    0.017886] NET: Registered protocol family 16
[    0.020381] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.024931] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.024944] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.025153] Serial: AMBA PL011 UART driver
[    0.026952] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.027408] uart-pl011 3f201000.serial: could not find pctldev for node /soc/gpio@7e200000/uart0_pins, deferring probe
[    0.058278] bcm2835-dma 3f007000.dma: DMA legacy API manager at a0813000, dmachans=0x1
[    0.059665] SCSI subsystem initialized
[    0.059888] usbcore: registered new interface driver usbfs
[    0.059952] usbcore: registered new interface driver hub
[    0.060190] usbcore: registered new device driver usb
[    0.070082] raspberrypi-firmware soc:firmware: Attached to firmware from 2018-04-16 18:16
[    0.071300] clocksource: Switched to clocksource arch_sys_counter
[    0.152082] VFS: Disk quotas dquot_6.6.0
[    0.152172] VFS: Dquot-cache hash table entries: 1024 (order 0, 4096 bytes)
[    0.152359] FS-Cache: Loaded
[    0.152560] CacheFiles: Loaded
[    0.161433] NET: Registered protocol family 2
[    0.162147] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
[    0.162213] TCP bind hash table entries: 4096 (order: 3, 32768 bytes)
[    0.162313] TCP: Hash tables configured (established 4096 bind 4096)
[    0.162417] UDP hash table entries: 256 (order: 1, 8192 bytes)
[    0.162449] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
[    0.162662] NET: Registered protocol family 1
[    0.163115] RPC: Registered named UNIX socket transport module.
[    0.163126] RPC: Registered udp transport module.
[    0.163135] RPC: Registered tcp transport module.
[    0.163143] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    0.221089] hw perfevents: enabled with armv7_cortex_a7 PMU driver, 7 counters available
[    0.224252] workingset: timestamp_bits=14 max_order=17 bucket_order=3
[    0.231923] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.232205] FS-Cache: Netfs 'nfs' registered for caching
[    0.232808] NFS: Registering the id_resolver key type
[    0.232845] Key type id_resolver registered
[... 170 lines cut ...]
[   12.208821] Bridge firewalling registered
[   12.245766] Bluetooth: Core ver 2.22
[   12.245882] NET: Registered protocol family 31
[   12.245893] Bluetooth: HCI device and connection manager initialized
[   12.245926] Bluetooth: HCI socket layer initialized
[   12.245946] Bluetooth: L2CAP socket layer initialized
[   12.245988] Bluetooth: SCO socket layer initialized
[   12.267740] Bluetooth: HCI UART driver ver 2.3
[   12.267763] Bluetooth: HCI UART protocol H4 registered
[   12.267775] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.268012] Bluetooth: HCI UART protocol Broadcom registered
[   12.660502] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[... cut ...]
[   15.129244] IPv6: ADDRCONF(NETDEV_UP): br-4b9a0ab1dd4a: link is not ready
[   21.092994] systemd-udevd[56]: starting version 215
[   21.201099] systemd-journald[49]: Received request to flush runtime journal from PID 1
[   38.869329] IPv6: ADDRCONF(NETDEV_UP): supervisor0: link is not ready
[   43.671401] Voltage normalised (0x00000000)
[   51.991422] Under-voltage detected! (0x00050005)
[   56.151367] Voltage normalised (0x00000000)
[   66.551349] Under-voltage detected! (0x00050005)
[   72.791337] Voltage normalised (0x00000000)

drewcovi, 2018-08-14:
@floion gonna try now with an older OS version... cant imagine what else is going wrong

drewcovi, 2018-08-15:
No dice 😢

shaunmulligan, 2018-10-10:
@drewcovi I assume the `dtoverlay=hyperpixel,hyperpixel-gpio-backlight` is not in resinOS default dtoverlays, is it included in the default set on raspbian yet? When I looked at it initially it always required manually copying that dtb into the `boot/overlays`

drewcovi, 2019-02-28:
@shaunmulligan you're 10% correct.

any reason custom overlays couldn't be somehow added to an application for preload?

shaunmulligan, 2019-03-01:
@drewcovi we can, but the problem we need to figure out is how to include them on hostOS updates too, as currently the overlays get updated with OS updates and we want to maintain that behaviour. We have some ideas on how to do this, but its still in the works. In the meantime i might try just get the hyperpixel4 dtb included in https://github.com/balena-os/balena-raspberrypi :)

dfranssen, 2019-04-04:
Not sure if it is relevant here, but after spending several hours getting the **touch** working for my hyperpixel**4** on a rpi3, it might be worth mentioning it somewhere :-)

In short: load the `goodix` driver with modprobe

Detailed version:
1. I also copied the `hyperpixel4.dtbo` to the `/boot/overlays` folder of the SD card flashed with balena-os (I built it from [source](https://github.com/pimoroni/hyperpixel4/tree/master/src) on a rpi3 with raspbian stretch lite, scp-ed it to my local machine and copied it to the other SD card)
2. configure the fleet variables in balenaCloud according the [CONFIG_LINES](https://github.com/pimoroni/hyperpixel4/blob/master/install.sh)
3. my docker is extending from `balenalib/%%BALENA_MACHINE_NAME%%-node:8` in order to run an electron/vuejs app via X11. I also copied the [hyperpixel4-init](https://github.com/pimoroni/hyperpixel4/blob/master/dist/hyperpixel4-init) to `/usr/bin/hyperpixel4-init` in the image. A running container will execute `CMD ["bash", "/usr/src/app/start.sh"]`
4.  the `start.sh` begins with `modprobe i2c-dev && modprobe goodix && /usr/bin/hyperpixel4-init`

@shaunmulligan @spanceac if it is relevant, the info could perhaps be added in the [documentation](https://www.balena.io/docs/learn/develop/hardware/i2c-and-spi) somewhere?

petrosagg, 2019-09-26:
@shaunmulligan @floion can we include the hyperpixel overlay to our images? It looks pretty popular

floion, 2019-09-27:
https://github.com/balena-os/balena-raspberrypi/pull/400

floion, 2019-09-27:
@shaunmulligan @petrosagg which ones to add? hyperpixel https://github.com/pimoroni/hyperpixel , hyperpixel4 https://github.com/pimoroni/hyperpixel4/ or both?

If for hyperpixel 4, for which combination as detailed in https://github.com/pimoroni/hyperpixel4/#manual-installation ?

shaunmulligan, 2019-09-27:
Hmm, @floion i think we should do the hyperpixel4, i wonder if their is a nice way to just have all four options with different names? some thing like:
- `hyperpixel4-pi3`
- `hyperpixel4-pi4`
- `hyperpixel4-square-pi3`
- `hyperpixel4-square-pi4`

Would that work?

floion, 2019-09-27:
Sure. Let's do it like that

floion, 2020-01-31:
Done in https://github.com/balena-os/balena-raspberrypi/pull/400 so closing this item

closed 2020-01-31 by floion