balena-os/balena-raspberrypi #226: build a development version of resinOS for the Raspberry Pi 3

thufuxu, 2018-08-19:
follow the guidance @ https://resinos.io/docs/custombuild/ but always failed at this step:

HostOS - CentOS7 Linux 3.10.0-862.9.1.el7.x86_64 

=====
ERROR: resin-image-1.0-r0 do_rootfs: /home/fuxu/resinOS/resin-raspberrypi/build/tmp/deploy/images/raspberrypi3/u-boot.bin is an invalid path referenced in RESIN_BOOT_PARTITION_FILES.
ERROR: resin-image-1.0-r0 do_rootfs: Function failed: resin_boot_dirgen_and_deploy (log file is located at /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_rootfs.99737)
ERROR: Logfile of failure stored in: /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/resin-image/1.0-r0/temp/log.do_rootfs.99737
ERROR: Task (/home/AAAA/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-core/images/resin-image.bb:do_rootfs) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3135 tasks of which 3134 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/AAAA/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-core/images/resin-image.bb:do_rootfs

ZubairLK, 2018-08-20:
@thufuxu looks like the u-boot recipe was not built. strange.
Can you share a snippet of your local.conf file?

```
zubairlk@zubair-xps-resin:~/resin/yocto/resin-raspberrypi3$ cat build/conf/local.conf  | grep -i u-boot -C5

# RPI BSP uses uncompressed kernel images by default
KERNEL_IMAGETYPE="zImage"
KERNEL_BOOTCMD="bootz"

# RPI Use u-boot. This needs to be 1 as we use u-boot
RPI_USE_U_BOOT = "1"

# Set this to 1 if development image is desired
DEVELOPMENT_IMAGE = "1"

zubairlk@zubair-xps-resin:~/resin/yocto/resin-raspberrypi3$ 
```

I was to know what is the value of `RPI_USE_U_BOOT`

thufuxu, 2018-08-20:
@ZubairLK Thanks. here is the response message:
======
$ cat build/conf/local.conf | grep -i u-boot -C5

# RPI BSP uses uncompressed kernel images by default
KERNEL_IMAGETYPE="zImage"
KERNEL_BOOTCMD="bootz"

# RPI Use u-boot. This needs to be 1 as we use u-boot
RPI_USE_U_BOOT = "1"

# Set this to 1 if development image is desired
DEVELOPMENT_IMAGE = "0"

thufuxu, 2018-08-20:
@ZubairLK I tried couple more times to build, the previous error (resin-image.bb:do_rootfs) disappeared. 

**but there are still another two errors existed to complete the image building**  message are as below.

My docker information:
====
$ docker version
Client:
 Version:           18.06.0-ce
 API version:       1.38
 Go version:        go1.10.3
 Git commit:        0ffa825
 Built:             Wed Jul 18 19:08:18 2018
 OS/Arch:           linux/amd64
 Experimental:      false

Server:
 Engine:
  Version:          18.06.0-ce
  API version:      1.38 (minimum version 1.12)
  Go version:       go1.10.3
  Git commit:       0ffa825
  Built:            Wed Jul 18 19:10:42 2018
  OS/Arch:          linux/amd64
  Experimental:     false

====build message
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.8"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "resin-os"
DISTRO_VERSION       = "2.14.3"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:da3625c52e1ab8985fba4fc3d133edf92142f182"
meta-oe              
meta-filesystems     
meta-networking      
meta-python          = "HEAD:a19aa29f7fa336cd075b72c496fe1102e6e5422b"
meta-raspberrypi     = "HEAD:f2e2a4376e84dec6e5efeabaa86a4406fddf7dcd"
meta-resin-common    
meta-resin-sumo      = "HEAD:6936d4603e71eb162edbb5d65c6027b80cf10a11"
meta-resin-raspberrypi = "master:9143c3be25b873bd5e8765d804f234650d24e988"
meta-rust            = "HEAD:3efa0f069ffe4dfcb3531ea89834e69cd2257157"

Initialising tasks: 100% |###############################################################################| Time: 0:00:02
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: docker-disk-resin_armv7hf-supervisor-v7.16.6-r0 do_compile: Function failed: do_compile (log file is located at /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/docker-disk/resin_armv7hf-supervisor-v7.16.6-r0/temp/log.do_compile.21664)
ERROR: Logfile of failure stored in: /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/raspberrypi3-poky-linux-gnueabi/docker-disk/resin_armv7hf-supervisor-v7.16.6-r0/temp/log.do_compile.21664
Log data follows:
| DEBUG: Executing shell function do_compile
| Sending build context to Docker daemon  311.6MB
| 
| Step 1/6 : FROM docker:17.06-dind
|  ---> 038dca6ce611
| Step 2/6 : RUN apk add --update util-linux shadow e2fsprogs && rm -rf /var/cache/apk/*
|  ---> Using cache
|  ---> ab6ca092422a
| Step 3/6 : ADD entry.sh /entry.sh
|  ---> Using cache
|  ---> ff81c7af316b
| Step 4/6 : RUN chmod a+x /entry.sh
|  ---> Using cache
|  ---> c26d69afb4b8
| Step 5/6 : VOLUME /resin-data
[... 38 lines cut ...]
| No package 'devmapper' found
| Removing bundles/
| 
| ---> Making bundle: dynbinary-balena (in bundles/dynbinary-balena)
| Building: bundles/dynbinary-balena/balena-17.12.0-dev
| Created binary: bundles/dynbinary-balena/balena-17.12.0-dev
| 
| # github.com/docker/docker/cmd/mobynit
| /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/recipe-sysroot-native/usr/lib/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
| gcc: error: unrecognized command line option ‘-no-pie’
| 
| WARNING: /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/temp/run.do_compile.21665:1 exit 2 from 'go build -ldflags '-extldflags "-static  -no-pie"' .'
| ERROR: Function failed: do_compile (log file is located at /home/AAAA/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/temp/log.do_compile.21665)
ERROR: Task (virtual:native:/home/AAAA/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-containers/balena/balena_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3763 tasks of which 3761 didn't need to be rerun and 2 failed.

Summary: 2 tasks failed:
  /home/AAAA/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-containers/docker-disk/docker-disk.bb:do_compile
  virtual:native:/home/AAAA/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-containers/balena/balena_git.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
[000000036][LOG]Build for raspberrypi3 failed. Check failed log in build/tmp/log/cooker/raspberrypi3 .
[000000036][LOG]If build for raspberrypi3 succeeded, final image should have been generated here:
[000000036][LOG]   build/tmp/deploy/images/raspberrypi3/resin-image-raspberrypi3.resinos-img
[000000036][LOG]Done.
[000000036][LOG]Done.

thufuxu, 2018-08-20:
$ docker info
Containers: 0
 Running: 0
 Paused: 0
 Stopped: 0
Images: 7
Server Version: 18.06.0-ce
Storage Driver: overlay2
 Backing Filesystem: xfs
 Supports d_type: true
 Native Overlay Diff: true
Logging Driver: json-file
Cgroup Driver: cgroupfs
Plugins:
 Volume: local
 Network: bridge host macvlan null overlay
 Log: awslogs fluentd gcplogs gelf journald json-file logentries splunk syslog
Swarm: inactive
Runtimes: runc
Default Runtime: runc
Init Binary: docker-init
containerd version: d64c661f1d51c48782c9cec8fda7604785f93587
runc version: 69663f0bd4b60df09991c08812a60108003fa340
init version: fec3683
Security Options:
 seccomp
  Profile: default
Kernel Version: 3.10.0-862.11.6.el7.x86_64
Operating System: CentOS Linux 7 (Core)
OSType: linux
Architecture: x86_64
CPUs: 8
Total Memory: 7.622GiB
Name: 192.168.75.131
ID: PHCH:H5CJ:ERKQ:IFIX:VBGF:B4NG:24M6:IPOD:USLR:LBGU:NW35:ZFCS
Docker Root Dir: /var/lib/docker
Debug Mode (client): false
Debug Mode (server): false
Username: thufuxu
Registry: https://index.docker.io/v1/
Labels:
Experimental: false
Insecure Registries:
 127.0.0.0/8
Live Restore Enabled: false

WARNING: bridge-nf-call-iptables is disabled
WARNING: bridge-nf-call-ip6tables is disabled

ZubairLK, 2018-08-20:
ResinOS by default needs docker configured with aufs.

For ubuntu, I installed the linux extra kernel package. Then configured docker daemon
with aufs by editing `/etc/docker/daemon.json`
```
{
  "storage-driver": "aufs"
}
```

And reloading/restarting docker services.

thufuxu, 2018-08-21:
@ZubairLK I made to install a kernel for Centos with aufs support, but there is one error left for the balena do_combile.

===
 $ grep aufs /proc/filesystems
nodev	aufs
===
$ docker info
Containers: 2
 Running: 0
 Paused: 0
 Stopped: 2
Images: 1
Server Version: 1.13.1
Storage Driver: aufs
 Root Dir: /var/lib/docker/aufs
 Backing Filesystem: xfs
 Dirs: 14
 Dirperm1 Supported: true
Logging Driver: json-file
Cgroup Driver: cgroupfs
Plugins: 
 Volume: local
 Network: bridge host macvlan null overlay
Swarm: inactive
Runtimes: runc
Default Runtime: runc
Init Binary: docker-init
containerd version:  (expected: aa8187dbd3b7ad67d8e5e3a15115d3eef43a7ed1)
runc version: 5eda6f6fd0c2884c2c8e78a6e7119e8d0ecedb77 (expected: 9df8b306d01f59d3a8029be411de015b7304dd8f)
init version: N/A (expected: 949e6facb77383876aeff8a6944dde66b3089574)
Security Options:
 seccomp
  Profile: default
Kernel Version: 4.18.3-1.el7.x86_64
Operating System: CentOS Linux 7 (Core)
OSType: linux
Architecture: x86_64
Number of Docker Hooks: 3
CPUs: 8
Total Memory: 7.765 GiB
Name: localhost.localdomain
ID: PHCH:H5CJ:ERKQ:IFIX:VBGF:B4NG:24M6:IPOD:USLR:LBGU:NW35:ZFCS
Docker Root Dir: /var/lib/docker
Debug Mode (client): false
Debug Mode (server): true
 File Descriptors: 15
 Goroutines: 25
 System Time: 2018-08-21T00:16:41.059923936-04:00
 EventsListeners: 0
Username: thufuxu
Registry: https://index.docker.io/v1/
WARNING: bridge-nf-call-iptables is disabled
WARNING: bridge-nf-call-ip6tables is disabled
Experimental: false
Insecure Registries:
 127.0.0.0/8
Live Restore Enabled: false
Registries: docker.io (secure)
===
ERROR: balena-native-17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0 do_compile: Function failed: do_compile (log file is located at /home/fuxu/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/temp/log.do_compile.16193)
ERROR: Logfile of failure stored in: /home/fuxu/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/temp/log.do_compile.16193
Log data follows:
| DEBUG: Executing shell function do_compile
| # WARNING! I don't seem to be running in a Docker container.
| # The result of this command might be an incorrect build, and will not be
| # officially supported.
| #
| # Try this instead: make all
| #
| 
| Package devmapper was not found in the pkg-config search path.
| Perhaps you should add the directory containing `devmapper.pc'
| to the PKG_CONFIG_PATH environment variable
| No package 'devmapper' found
| Removing bundles/
| 
| ---> Making bundle: dynbinary-balena (in bundles/dynbinary-balena)
| Building: bundles/dynbinary-balena/balena-17.12.0-dev
| Created binary: bundles/dynbinary-balena/balena-17.12.0-dev
| 
| # github.com/docker/docker/cmd/mobynit
| /home/fuxu/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/recipe-sysroot-native/usr/lib/go/pkg/tool/linux_amd64/link: running gcc failed: exit status 1
| gcc: error: unrecognized command line option ‘-no-pie’
| 
| WARNING: /home/fuxu/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/temp/run.do_compile.16193:1 exit 2 from 'go build -ldflags '-extldflags "-static  -no-pie"' .'
| ERROR: Function failed: do_compile (log file is located at /home/fuxu/resinOS/resin-raspberrypi/build/tmp/work/x86_64-linux/balena-native/17.12.0-dev+git2fe3ad1568c1b783a9201fc3082452ad79d7f396-r0/temp/log.do_compile.16193)
ERROR: Task (virtual:native:/home/fuxu/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-containers/balena/balena_git.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3771 tasks of which 3764 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/home/fuxu/resinOS/resin-raspberrypi/build/../layers/meta-resin/meta-resin-common/recipes-containers/balena/balena_git.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
[000000046][LOG]Build for raspberrypi3 failed. Check failed log in build/tmp/log/cooker/raspberrypi3 .
[000000046][LOG]If build for raspberrypi3 succeeded, final image should have been generated here:
[000000046][LOG]   build/tmp/deploy/images/raspberrypi3/resin-image-raspberrypi3.resinos-img
[000000046][LOG]Done.

closed 2018-08-23 by thufuxu