balena-os/balena-raspberrypi #232: can't boot correctly with the built image

thufuxu, 2018-08-24:
I built the image following the https://resinos.io/docs/custombuild/#bake-your-own-image.

using the tagged version 2.13.6.

looks like it succeeded.  then I use the Etcher in Windows to flash the built image to SD card. but when booting, the RPI stopped at the yellow logo screen and can't boot correctly.

**Anyone know why?**

PS: if I flash (also using the Etcher) the downloaded image from resinos.io, it can boot correctly.

PS: the built image and the downloaded image have the same size. 1.68 GB (1,807,745,024 bytes)

floion, 2018-08-26:
@thufuxu do you have an USB to TTL serial cable? You can use such a cable on a development image so that you can inspect the problem over the serial console

thufuxu, 2018-08-27:
the issue was solved... looks it is because that the auto-generated local.conf didn't include configure some variables. After modified them, the issue disappeared. 
```
# RPI BSP uses uncompressed kernel images by default
KERNEL_IMAGETYPE="zImage"
KERNEL_BOOTCMD="bootz"

# RPI Use u-boot. This needs to be 1 as we use u-boot
RPI_USE_U_BOOT = "1"

```

closed 2018-08-27 by thufuxu