balena-os/balena-raspberrypi #238: Error response from daemon: No such image: certificates

thufuxu, 2018-09-05:
```
ERROR: mkfs-hostapp-native-1.0-r0 do_compile: Function failed: do_compile (log file is located at /home/fuxu/resinOS/qemu/build/tmp/work/x86_64-linux/mkfs-hostapp-native/1.0-r0/temp/log.do_compile.105951)
ERROR: Logfile of failure stored in: /home/fuxu/resinOS/qemu/build/tmp/work/x86_64-linux/mkfs-hostapp-native/1.0-r0/temp/log.do_compile.105951
Log data follows:
| DEBUG: Executing shell function do_compile
| The command '/bin/sh -c apt-get update && apt-get install -y 	ca-certificates 	iptables' returned a non-zero code: 100
| Error response from daemon: No such image: certificates
| WARNING: /home/fuxu/resinOS/qemu/build/tmp/work/x86_64-linux/mkfs-hostapp-native/1.0-r0/temp/run.do_compile.105951:1 exit 1 from 'DOCKER_API_VERSION=1.22 docker save "$IMAGE_ID" > /home/fuxu/resinOS/qemu/build/tmp/work/x86_64-linux/mkfs-hostapp-native/1.0-r0/work/mkfs-hostapp-ext4-image.tar'
| ERROR: Function failed: do_compile (log file is located at /home/fuxu/resinOS/qemu/build/tmp/work/x86_64-linux/mkfs-hostapp-native/1.0-r0/temp/log.do_compile.105951)
ERROR: Task (/home/fuxu/resinOS/qemu/build/../layers/meta-resin/meta-resin-common/recipes-containers/mkfs-hostapp-native/mkfs-hostapp-native.bb:do_compile) failed with exit code '1'
```

fmarion, 2018-10-09:
Got the exact same issue building resin-beaglebone. 
@thufuxu have you found a way to fix this ?

thufuxu, 2018-10-12:
not really. The 2.13.6 is the only version I built successfully so far.@fmarion

agherzan, 2018-10-12:
I hit the same issue yesterday. And this was due to the fact the the vpn connection I had was hung. So the server had no Internet access. So I suspect that this is due to network issues when running a container.

closed 2018-11-01 by thufuxu