balena-os/balena-raspberrypi #240: Update firmware to support rPI 3B+ PoE hat (+ fan)

christopherhex, 2018-09-06:
New Raspberry [Pi PoE hat](
https://www.raspberrypi.org/products/poe-hat/) has a fan that can be controlled through the firmware. This requires an updated linux-firmware.  

Also see this commit:  [raspberrypi/firmware - 953a33](https://github.com/raspberrypi/firmware/commit/953a33962f0e3a33f7627248f47b8378cb40e6e4)

floion, 2018-09-09:
Thanks for the input. Andrei already created https://github.com/agherzan/meta-raspberrypi/pull/312 to handle this in the BSP. Let's see if this gets backported to Sumo

agherzan, 2018-09-11:
Pushed a PR which adds this support in sumo: https://github.com/agherzan/meta-raspberrypi/pull/314

christopherhex, 2018-09-12:
Hi @agherzan, thanks! 
Any ETA when it will be included in an upcoming release of this repo?

agherzan, 2018-09-12:
Keep an eye on the releases for the RaspberryPi. I'd probably say 2 weeks.

floion, 2018-09-13:
Implemented in https://github.com/resin-os/resin-raspberrypi/pull/243

closed 2018-11-16 by floion