balena-os/balena-raspberrypi #246: Build fail (patch failed: drivers/block/loop.c)

ruudboon, 2018-09-28:
Hi,

I'm trying to build the latest ResinOS for the RPI but unable to do so.
I tried doing a clean build also but didn't work out.
```
DEBUG: Python function do_kernel_resin_aufs_fetch_and_unpack finished
DEBUG: Executing shell function apply_aufs_patches
error: patch failed: drivers/block/loop.c:686
Falling back to three-way merge...
Applied patch to 'drivers/block/loop.c' with conflicts.
U drivers/block/loop.c
WARNING: exit code 1 from a shell command.
ERROR: Function failed: apply_aufs_patches (log file is located at /home/user/resinbuild/build/tmp/work/raspberrypi3-poky-linux-gnueabi/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/temp/log.do_kernel_resin_aufs_fetch_and_unpack.13443)
```

agherzan, 2018-10-01:
Can you share the exact revision you are trying to build? Do you have the submodule updated?

ruudboon, 2018-10-01:
@agherzan I've used master from september 19th. release_v2.18.0+rev1. Clean checkout and git submodule update --init --recursive. Some other members had the same problems (see gitter).
I will give the latest 2.19 a shot now.

ruudboon, 2018-10-01:
Getting different errors now
```
 In file included from /home/ruud/resinos-raspberrypi-github/build/tmp/work-shared/raspberrypi3/kernel-source/fs/aufs/aufs.h:45:0,
|                  from /home/ruud/resinos-raspberrypi-github/build/tmp/work-shared/raspberrypi3/kernel-source/fs/aufs/module.c:24:
| /home/ruud/resinos-raspberrypi-github/build/tmp/work-shared/raspberrypi3/kernel-source/fs/aufs/file.h: In function 'au_vm_prfile_set':
| /home/ruud/resinos-raspberrypi-github/build/tmp/work-shared/raspberrypi3/kernel-source/fs/aufs/file.h:332:7: error: 'struct vm_area_struct' has no member named 'vm_prfile'; did you mean 'vm_file'?
|   vma->vm_prfile = file;
|        ^~~~~~~~~
|        vm_file
|   CC      mm/madvise.o
```

floion, 2018-10-02:
Hi @ruudboon , can you paste here the exact commands you used to build it? Including from the very first ones (git repository checkout) up to the barys build line
Thank you

ruudboon, 2018-10-04:
@floion of course
```
export LANGUAGE=en_US.UTF-8
export LANG=en_US.UTF-8
export LC_ALL=en_US.UTF-8
git clone https://github.com/resin-os/resin-raspberrypi
cd resin-raspberrypi
git submodule update --init --recursive
./resin-yocto-scripts/build/barys -m raspberrypi3
```

Rebuild
```
export LANGUAGE=en_US.UTF-8
export LANG=en_US.UTF-8
export LC_ALL=en_US.UTF-8
cd resin-raspberrypi
git pull
git submodule update --init --recursive
```

ruudboon, 2018-10-04:
I've did a clean checkout and builded again. Looks like it's working now.
I did see some warnings but no errors this time!
```
WARNING: bzip2-native-1.0.6-r5 do_fetch: Checksum mismatch for local file /home/ruud/resin-raspberrypi/build/downloads/bzip2-1.0.6.tar.gz
Cleaning and trying again.
WARNING: bzip2-native-1.0.6-r5 do_fetch: Renaming /home/ruud/resin-raspberrypi/build/downloads/bzip2-1.0.6.tar.gz to /home/ruud/resin-raspberrypi/build/downloads/bzip2-1.0.6.tar.gz_bad-checksum_00ef481897e86efa3fa00769723c790f
WARNING: bzip2-native-1.0.6-r5 do_fetch: Checksum failure encountered with download of http://www.bzip.org/1.0.6/bzip2-1.0.6.tar.gz - will attempt other sources if available
WARNING: linux-raspberrypi-1_4.14.68+gitAUTOINC+8c8666ff6c-r0 do_kernel_resin_checkconfig: Checking for CONFIG_IP_NF_TARGET_LOG=m in the kernel configs failed for misc.
WARNING: lsof-4.89-r0 do_fetch: Failed to fetch URL http://www.mirrorservice.org/sites/lsof.itap.purdue.edu/pub/tools/unix/lsof/lsof_4.89.tar.bz2, attempting MIRRORS if available

```

closed 2018-10-04 by ruudboon