balena-os/balena-raspberrypi #261: error: no matching package named `atty` found (required by `cargo`)

thufuxu, 2018-11-01:
`ERROR: cargo-native-0.25.0-r0 do_compile: Function failed: do_compile (log file is located at /home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.69181)
ERROR: Logfile of failure stored in: /home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.69181
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: cargo = /home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo
| Usage: which [options] [--] COMMAND [...]
| Write the full path of COMMAND(s) to standard output.
| 
|   --version, -[vV] Print version and exit successfully.
|   --help,          Print this help and exit successfully.
|   --skip-dot       Skip directories in PATH that start with a dot.
|   --skip-tilde     Skip directories in PATH that start with a tilde.
|   --show-dot       Don't expand a dot to current directory in output.
|   --show-tilde     Output a tilde for HOME directory for non-root.
|   --tty-only       Stop processing options on the right if not on tty.
|   --all, -a        Print all matches in PATH, not just the first
|   --read-alias, -i Read list of aliases from stdin.
|   --skip-alias     Ignore option --read-alias; don't read stdin.
|   --read-functions Read shell functions from stdin.
|   --skip-functions Ignore option --read-functions; don't read stdin.
| 
| Recommended use is to write the output of (alias; declare -f) to standard
| input, so that which can show aliases and shell functions. See which(1) for
| examples.
| 
| If the options --read-alias and/or --read-functions are specified then the
| output can be a full alias or function definition, optionally followed by
| the full path of each command used inside of those.
| 
| Report bugs to <which-bugs@gnu.org>.
| NOTE: rustc =
| NOTE: /home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo build -v --target x86_64-linux --release
| error: no matching package named `atty` found (required by `cargo`)
| location searched: registry `https://github.com/rust-lang/crates.io-index`
| version required: = 0.2.11
| WARNING: /home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/run.do_compile.69181:1 exit 101 from '"/home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.24.0-x86_64-unknown-linux-gnu/cargo/bin/cargo" build -v --target x86_64-linux --release "$@"'
| ERROR: Function failed: do_compile (log file is located at /home/user/balenaOS/rpi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.69181)
`

siredmar, 2019-03-04:
@thufuxu what did you do to resolve this problem?
I have exact the same

thufuxu, 2019-03-04:
@siredmar it’s because of network issue usually. I setup a mirror for crates, then solved.

siredmar, 2019-03-05:
What do you mean exactly by saying network issue?
And did you change the crate URIs from the recepies to your mirror?

thufuxu, 2019-03-05:
network issue means that your network may not be able to download the crates listed here. You can try downloading again or setup proxy or setup a mirror of the crates.io.

siredmar, 2019-03-05:
The system was able to download the crates as they were present in the download section of bitbake. I tried several times, also forcing another version of atty (e.g. 0.2.11 instead of 0.2.6)

floion, 2019-03-05:
@siredmar so did they download work after all?

siredmar, 2019-03-05:
@floion , yes they did. 

```
$ ls /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/bitbake/atty-0.2.6 
appveyor.yml  Cargo.toml  Cargo.toml.orig  CHANGELOG.md  examples  LICENSE  README.md  rustfmt.toml  src

```

Also the crate used in layers/meta-rust/recipes-devtools/cargo/cargo_0.25.0.bb is atty-0.2.6

```

SRC_URI += " \
...
crate://crates.io/atty/0.2.6 \

```

I suggest you try a clean build for yourself and see what happens. 
I cloned the repo, pulled submodules and called

```
$ ./balena-yocto-scripts/build/barys -m raspberrypi3
```

siredmar, 2019-03-05:
Kind of pinning down the problem here. 
I found out that atty is not the problem here. No package can be fetched with cargo. I tried that by using the …/cargo-native/0.25.0-r0/temp/run.do_compile scripts and modifying it to build a minimalist example using only one dependency.

What i also found out was that the cargo_home is not 

```
balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo_home/ 
```
but
```
balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/cargo_home/
```
Inside is a config file that contains:

```
...
[source.bitbake]
directory = "/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/cargo_home/bitbake"
[source.crates-io]
replace-with = "bitbake"
local-registry = "/nonexistant"
...
```
This file comes from the meta-rust layer (https://github.com/meta-rust/meta-rust/blob/f2f17c58b00d12e795fc9c5f66c1bb82759bf72f/classes/cargo_common.bbclass#L46)
This change came in in this commit https://github.com/meta-rust/meta-rust/commit/48e2a7db2588d55ee5c43cf5242ad41aa16a2c1c#diff-86fe95c88946300256cccb34400231e7

However commenting those lines out downloads the crates in the ../git/cargo_home 

@thufuxu you said you mirrored the crate server. Did you came accross anything i found out here?

siredmar, 2019-03-05:
Update: I succeeded in building the balena-raspberry image (not tested if it actually boots up) with the current master on meta-rust layer (https://github.com/meta-rust/meta-rust/tree/0621067c560f6e3e1a4c4c08ac173393bbafad5c) and removing those files as the version of cargo is now different:
```
balena-raspberrypi/layers/meta-resin/meta-resin-sumo/recipes-devtools/cargo/cargo_0.25.0.bbappend 

balena-raspberrypi/layers/meta-resin/meta-resin-sumo/recipes-devtools/cargo/files/Cargo.lock file
```

floion, 2019-03-05:
I just tried on my development machine and was able to have it finish without this error for the same machine as you did:

> ./balena-yocto-scripts/build/barys -m raspberrypi2

[000000004][LOG]Run build for raspberrypi2: MACHINE=raspberrypi2 bitbake resin-image 
[000000004][LOG]This might take a while ...
Parsing recipes: 100% |#############################################################################################################################################################################| Time: 0:00:35
Parsing of 2152 .bb files complete (0 cached, 2152 parsed). 3049 targets, 314 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi2"
DISTRO               = "balena-os"
DISTRO_VERSION       = "2.30.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "HEAD:24a833f40d837407baa4e8192f372d998a776b6c"
meta-oe              
meta-filesystems     
meta-networking      
meta-python          = "HEAD:be79b8b111a968efdbe5e1482d0c246d0b24763e"
meta-raspberrypi     = "HEAD:2d40b000021bc8a9ef7f329ed0ad410f8d227b97"
meta-resin-common    
meta-resin-sumo      = "HEAD:1aba4a4a644f74a80636b98e70407c06c8db6f33"
meta-resin-raspberrypi = "master:5ec5540bbc3d67b86f9b0c8a58da3939c9bac700"
meta-rust            = "HEAD:3efa0f069ffe4dfcb3531ea89834e69cd2257157"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.2/x86_64-nativesdk-libc.tar.bz2;sha256sum=e3b77208169bf1ac4e89496f3cdbf27695f5b18a2694a908a793390f28b67f83
Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:03
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: bzip2-native-1.0.6-r5 do_fetch: Checksum mismatch for local file /home/floion/builds/teste_headers/balena-raspberrypi/build/downloads/bzip2-1.0.6.tar.gz
Cleaning and trying again.
WARNING: bzip2-native-1.0.6-r5 do_fetch: Renaming /home/floion/builds/teste_headers/balena-raspberrypi/build/downloads/bzip2-1.0.6.tar.gz to /home/floion/builds/teste_headers/balena-raspberrypi/build/downloads/bzip2-1.0.6.tar.gz_bad-checksum_f4b5249c17053f5c4c85d870feef4f48
WARNING: bzip2-native-1.0.6-r5 do_fetch: Checksum failure encountered with download of http://www.bzip.org/1.0.6/bzip2-1.0.6.tar.gz - will attempt other sources if available
WARNING: expat-native-2.2.5-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/expat/expat-2.2.5.tar.bz2, attempting MIRRORS if available
WARNING: freetype-2.9-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/freetype/freetype-2.9.tar.bz2, attempting MIRRORS if available
WARNING: libusb-compat-1_0.1.5-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/libusb/libusb-compat-0.1.5.tar.bz2, attempting MIRRORS if available
WARNING: tcl-native-8.6.8-r0 do_fetch: Failed to fetch URL http://downloads.sourceforge.net/tcl/tcl8.6.8-src.tar.gz, attempting MIRRORS if available
WARNING: lsof-4.89-r0 do_fetch: Failed to fetch URL http://www.mirrorservice.org/sites/lsof.itap.purdue.edu/pub/tools/unix/lsof/lsof_4.89.tar.bz2, attempting MIRRORS if available
WARNING: linux-raspberrypi-1_4.14.98+gitAUTOINC+5d63a4595d-r0 do_kernel_resin_checkconfig: Checking for CONFIG_IP_NF_TARGET_LOG=m in the kernel configs failed for misc.
NOTE: Tasks Summary: Attempted 3827 tasks of which 10 didn't need to be rerun and all succeeded.

Summary: There were 9 WARNING messages shown.
[000011361][LOG]Build for raspberrypi2 suceeded.
[000011361][LOG]If build for raspberrypi2 succeeded, final image should have been generated here:
[000011361][LOG]   build/tmp/deploy/images/raspberrypi2/resin-image-raspberrypi2.resinos-img
[000011361][LOG]Done.

siredmar, 2019-03-06:
@floion now i am completely confused. Did you clone the repo freshly with all its submodules?
What DNS server did you configure in your host?
What OS are you building with?

I am running Ubuntu 18.04 - and getting this warning 
**WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.**
My primary DNS Server is google 8.8.8.8
Network connection is stable and not running through a proxy or VPN.

What does your **balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/cargo_home/config** look like?

floion, 2019-03-06:
The build was done in a fresh clone (with `git clone --recursive`)

> cat /etc/resolv.conf 

```
nameserver 213.133.98.98
nameserver 213.133.99.99
nameserver 213.133.100.100
nameserver 2a01:4f8:0:a102::add:9999
nameserver 2a01:4f8:0:a0a1::add:1010
nameserver 2a01:4f8:0:a111::add:9898
```

I built on a 16.04.2 LTS (Xenial Xerus).

> cat build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/cargo_home/config

```
paths = [
]
[source.bitbake]
directory = "/home/floion/builds/teste_headers/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git/cargo_home/bitbake"
[target.x86_64-linux]
linker = '/home/floion/builds/teste_headers/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-ccld'
```

siredmar, 2019-03-06:
Your cargo_home/config file does look different. Mine does have the section in it:
```
[source.crates-io]
replace-with = "bitbake"
local-registry = "/nonexistant"
``` 

Do you have any idea why? Could it be because i am building with Ubuntu 18.04?

floion, 2019-03-06:
Can't say I know why it looks different for you.
Would you have a 16.04 to build on? @spanceac did you manage to get the build going on 18.04?

siredmar, 2019-03-07:
@floion i did a clean build on a new setup vanilla Ubuntu 16.04.06 and it was running without issues.

floion, 2019-03-07:
Great. So I guess 18.04 will need to get official support from Yocto until this can be fixed on that OS version