balena-os/balena-raspberrypi #262: boot failed using own built image

thufuxu, 2018-11-03:
I used the 2.15.1+rev2 to build the resinOS image. 
I didn't modify anything after GIT from the github.
Generally, the building process is smooth and the result is succeed with below notice.
`NOTE: Tasks Summary: Attempted 3784 tasks of which 3784 didn't need to be rerun and all succeeded.
[000000021][LOG]Build for raspberrypi3 suceeded.
[000000021][LOG]If build for raspberrypi3 succeeded, final image should have been generated here:
[000000021][LOG]   build/tmp/deploy/images/raspberrypi3/resin-image-raspberrypi3.resinos-img
`
Then I use the resin-cli to flash the image to the SD card. Unfortunately, I can't boot the RPI3 with the built image (2.15.1+rev2 version). I mean: the boot stopped at the LOGO screen, no further step.

At the first beginning of the boot, there is some message:

`found U-Boot scrpt /boot.scr

** Unable to read file resin-image-flasher **

** Bad device mmc 1 **
** Bad device mmc 2 **
** Bad device usb 0 **
** Bad device usb 1 **
** Bad device usb 2 **

** Unable to read file resinOS——uEnv.txt**
`

Where I should fix?

thufuxu, 2018-11-03:
By the way, using the same SD card, I can flash the download official resinOS image, and it can boot.

floion, 2018-11-07:
Hi. Those messages should not present a problem.
Did you try to build a development image so that you can login over the serial console?
You can build a development build by using the `-d` flag to barys. Just make sure this is on a new build

thufuxu, 2018-11-08:
@floion I didn't build development iamge but a production image.

ZubairLK, 2018-11-08:
@thufuxu can you elaborate boot stopped at logo screen? Do you see the yellow resin logo? And why you are thinking this boot 'failed'

If I remember correctly, the production image is locked tightly. And that yellow logo is it.
The image you generated is a blank image and not tied to your balena.io account. So you'll need to inject your applications config.json into the image. Unless you inject a config.json into the image, it won't connect to a resin.io/balena.io/balenaCloud dashboard.. 

And if a config.json exists, internet connectivity exists, then an application is downloaded and run. Until that application tries to mess with the display, the yellow logo will continue to display.

thufuxu, 2018-11-09:
@ZubairLK Thanks. This makes me clear. then it should not a boot fail.

closed 2018-11-09 by thufuxu