balena-os/balena-raspberrypi #265: Unable to build image due to sd8887-mrvl git repo not being available

pjb304, 2018-11-06:
When trying to build a new image of this OS I get the following error 

`ERROR: sd8887-mrvl-1.0-r0 do_fetch: Fetcher failure for URL: 'git://git@github.com/resin-io/sd8887-mrvl.git;protocol=ssh;branch=raspbian-build'. Unable to fetch URL from any source.`

Also when trying to browse the repo on the github webpage I get a 404 error.  Has this repo been moved / deleted as part of the rebrand?  I can provide the full log file if that would be helpful.

pjb304, 2018-11-06:
The configuration for the build that failed was
```Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "fincm3"
DISTRO               = "balena-os"
DISTRO_VERSION       = "2.26.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky            = "HEAD:24a833f40d837407baa4e8192f372d998a776b6c"
meta-oe
meta-filesystems
meta-networking
meta-python          = "HEAD:be79b8b111a968efdbe5e1482d0c246d0b24763e"
meta-raspberrypi     = "HEAD:2d40b000021bc8a9ef7f329ed0ad410f8d227b97"
meta-resin-common
meta-resin-sumo      = "HEAD:426f69000bf1c6f632345a1b63a1572f99c55e77"
meta-resin-raspberrypi = "master:f17ed744c82cce11883825a0457b8e9028449c20"
meta-rust            = "HEAD:3efa0f069ffe4dfcb3531ea89834e69cd2257157"
```

floion, 2018-11-07:
Hi @pjb304 thanks for the report. Indeed, this was due to the rename
Fixed by https://github.com/balena-os/resin-raspberrypi/pull/266

floion, 2018-11-16:
Hi @pjb304 that is a private repository, that's why you cannot access it.
Are you building for rpi3?

pjb304, 2018-11-16:
Hi, yes I'm building for a Pi 3 following the instructions from https://github.com/balena-os/resin-raspberrypi which state to run `./resin-yocto-scripts/build/barys` which I would expect to be able to run without errors as no warnings that they are to be expected are given in the instructions.

floion, 2019-02-28:
@pjb304 can this be closed now?

pjb304, 2019-02-28:
I see you have made the repo public now which has had the effect of downgrading the error to a warning

`WARNING: sd8887-mrvl-1.0-r0 do_fetch: Failed to fetch URL git://git@github.com/balena-io/sd8887-mrvl.git;protocol=ssh;branch=master, attempting MIRRORS if available
`
It looks like it then managed to fetch the file from a mirror (presumably using the https URL).

Before closing it would be nice to fix the warning.

agherzan, 2019-02-28:
@floion we need to switch to https to silence this warning.

agherzan, 2019-11-21:
@floion @ZubairLK github.com/balena-io/sd8887-mrvl returns 404. Was that taken off for public access?

floion, 2019-11-21:
Yes. We need to add the binary driver instead. We'll do this in the following days

alexgg, 2021-03-04:
Binary drivers are already available. We will also soon move to using a GPLed version of the driver, see https://github.com/balena-os/balena-raspberrypi/issues/619.

closed 2021-03-04 by alexgg