balena-os/balena-raspberrypi #268: raspberry3-64 build fails - No rule to make target 'zImage'

pjb304, 2018-11-07:
When compiling the raspberry3-64 version of the OS with the following versions 
```Build Configuration:
BB_VERSION           = "1.38.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi3-64"
DISTRO               = "balena-os"
DISTRO_VERSION       = "2.26.0"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta
meta-poky            = "HEAD:24a833f40d837407baa4e8192f372d998a776b6c"
meta-oe
meta-filesystems
meta-networking
meta-python          = "HEAD:be79b8b111a968efdbe5e1482d0c246d0b24763e"
meta-raspberrypi     = "HEAD:2d40b000021bc8a9ef7f329ed0ad410f8d227b97"
meta-resin-common
meta-resin-sumo      = "HEAD:426f69000bf1c6f632345a1b63a1572f99c55e77"
meta-resin-raspberrypi = "master:f17ed744c82cce11883825a0457b8e9028449c20"
meta-rust            = "HEAD:3efa0f069ffe4dfcb3531ea89834e69cd2257157"
```
I get an error and build failure
```
ERROR: linux-raspberrypi-1_4.14.68+gitAUTOINC+8c8666ff6c-r0 do_compile: oe_runmake failed
ERROR: linux-raspberrypi-1_4.14.68+gitAUTOINC+8c8666ff6c-r0 do_compile: Function failed: do_compile (log file is located at /home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/temp/log.do_compile.35028)
ERROR: Logfile of failure stored in: /home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/temp/log.do_compile.35028
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 HOSTCC=gcc  -isystem/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/usr/lib -L/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native/lib -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/pjb1u12/resin-raspberrypi/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 HOSTCPP=gcc  -E zImage CC=aarch64-poky-linux-gcc   -fuse-ld=bfd -fdebug-prefix-map=/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0=/usr/src/debug/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0 -fdebug-prefix-map=/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot= -fdebug-prefix-map=/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot-native=  -fdebug-prefix-map=/home/pjb1u12/resin-raspberrypi/build/tmp/work-shared/raspberrypi3-64/kernel-source=/usr/src/kernel  --sysroot=/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot -ffile-prefix-map=/home/pjb1u12/resin-raspberrypi/build/tmp/work-shared/raspberrypi3-64/kernel-source=/kernel-source/  LD=aarch64-poky-linux-ld.bfd    --sysroot=/home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/recipe-sysroot LOADADDR=0x00008000
| make[2]: *** No rule to make target 'zImage'.  Stop.
| Makefile:146: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:24: recipe for target '__sub-make' failed
| make: *** [__sub-make] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/pjb1u12/resin-raspberrypi/build/tmp/work/raspberrypi3_64-poky-linux/linux-raspberrypi/1_4.14.68+gitAUTOINC+8c8666ff6c-r0/temp/log.do_compile.35028)
```

floion, 2018-11-09:
Hi. Currently we do not maintain the 64 bits build

floion, 2019-02-28:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/297

closed 2019-02-28 by floion