balena-os/balena-raspberrypi #274: Add support for the Raspberry Pi 3 A+ board

imrehg, 2018-11-28:
This needs most likely 4.14.79 kernel + firmware. Accelerating it to be added to the current setup, not waiting for new yocto version.

[Flowdock convo](https://www.flowdock.com/app/rulemotion/pub/threads/7I8z6rLqeACwLsEqyka3YJhaMC4)

Timmoz, 2018-12-17:
Any update on this? Is someone working on a RPI 3A+ support?

floion, 2018-12-17:
Hi. The work for this has been done and merged in our staging environment. The next rpi3 release (which is due soon, a week or so, maybe less) will be compatible with the 3a+

Timmoz, 2018-12-17:
Great. I'll wait an other week, no problem.

Jaeyoung-Lim, 2018-12-30:
Are there any updates regarding this?

Jaeyoung-Lim, 2018-12-30:
@floion It would be helpful if we can get a ETA to prepare, Thanks

paulchambers, 2019-01-01:
@imrehg 2.29.0+rev1 works on a Raspberry Pi 3 A+ for me :)

    SystemTime:        2019-01-01T12:32:24.121336033Z
    KernelVersion:     4.14.79
    OperatingSystem:   balenaOS 2.29.0+rev1
    Architecture:      armv7l

imrehg, 2019-01-01:
Support added in 2.29.0+rev1
https://github.com/balena-os/balena-raspberrypi/blob/master/CHANGELOG.md#v2290rev1

closed 2019-01-01 by imrehg