balena-os/balena-raspberrypi #275: kernel panics with reboot on OS v2.27.0

shaunmulligan, 2018-11-29:
When issuing a reboot from the serial console on a .dev v2.27.0 version of the OS on raspberry pi 3 B+, i see the following kernel panic and then when the watchdog kicks in, the device actually reboots:

```
[  167.632713] ------------[ cut here ]------------
[  167.637525] WARNING: CPU: 1 PID: 0 at /usr/src/kernel/kernel/rcu/tree.c:2725 rcu_process_callbacks+0x654/0x670
[  167.647824] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic ipt_MASQUERADE nf_nat_masquerade_ipv4 br_netfilter nf_conntrack_netlink nfnetlink xt_owner i2c_dev brcmfmac l
[  167.677488] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G         C      4.14.68 #1
[  167.684854] Hardware name: BCM2835
[  167.688334] [<80110c5c>] (unwind_backtrace) from [<8010cbc0>] (show_stack+0x20/0x24)
[  167.696239] [<8010cbc0>] (show_stack) from [<8085ed98>] (dump_stack+0xcc/0x110)
[  167.703700] [<8085ed98>] (dump_stack) from [<8011e950>] (__warn+0xf8/0x110)
[  167.710805] [<8011e950>] (__warn) from [<8011ea80>] (warn_slowpath_null+0x30/0x38)
[  167.718531] [<8011ea80>] (warn_slowpath_null) from [<8018932c>] (rcu_process_callbacks+0x654/0x670)
[  167.727766] [<8018932c>] (rcu_process_callbacks) from [<801016c4>] (__do_softirq+0x164/0x3f0)
[  167.736469] [<801016c4>] (__do_softirq) from [<80124b20>] (irq_exit+0x100/0x174)
[  167.744019] [<80124b20>] (irq_exit) from [<8017ad44>] (__handle_domain_irq+0x70/0xc4)
[  167.752010] [<8017ad44>] (__handle_domain_irq) from [<8010155c>] (bcm2836_arm_irqchip_handle_irq+0xac/0xb0)
[  167.764937] [<8010155c>] (bcm2836_arm_irqchip_handle_irq) from [<8087bf3c>] (__irq_svc+0x5c/0x7c)
[  167.779953] Exception stack(0xbcd2bf38 to 0xbcd2bf80)
[  167.788053] bf20:                                                       00000000 004e1b00
[  167.802271] bf40: 3d1ce000 00000000 ffffe000 81404e10 81404dac 8148d9f2 80a7ab30 00000001
[  167.816534] bf60: 00000000 bcd2bf94 bcd2bf88 bcd2bf88 801090cc 801090d0 60000013 ffffffff
[  167.830925] [<8087bf3c>] (__irq_svc) from [<801090d0>] (arch_cpu_idle+0x34/0x4c)
[  167.844638] [<801090d0>] (arch_cpu_idle) from [<8087b69c>] (default_idle_call+0x34/0x48)
[  167.859087] [<8087b69c>] (default_idle_call) from [<80166db0>] (do_idle+0xf0/0x158)
[  167.873278] [<80166db0>] (do_idle) from [<801670d8>] (cpu_startup_entry+0x28/0x2c)
[  167.887423] [<801670d8>] (cpu_startup_entry) from [<8010eaac>] (secondary_start_kernel+0x160/0x16c)
[  167.903095] [<8010eaac>] (secondary_start_kernel) from [<001019ec>] (0x1019ec)
[  167.913654] ---[ end trace 8865936c37c2a57e ]---

shaunmulligan, 2018-12-06:
Saw this again today and it seems like the watchdog never kicked in either. My device is just hung trying to reboot. Its a .dev version from production and hasn't ever connected to wifi yet, just have serial console connected up. cc @ZubairLK , @floion  and @agherzan 

logs:
```
root@7eae28f:~# reboot
[  412.134501] Unable to handle kernel NULL pointer dereference at virtual address 00000004
[  412.142819] pgd = bb530000
[  412.145581] [00000004] *pgd=00000000
[  412.149254] Internal error: Oops: 5 [#1] SMP ARM
[  412.153970] Modules linked in: ipt_REJECT nf_reject_ipv4 bnep hci_uart btbcm serdev bluetooth ecdh_generic ipt_MASQUERADE nf_nat_masquerade_ipv4 br_netfilter nf_conntrack_netlink nfnetlinkl
[  412.185887] CPU: 3 PID: 1737 Comm: umount Tainted: G         C      4.14.68 #1
[  412.185895] Hardware name: BCM2835
[  412.185899] task: bbcacb00 task.stack: b399a000
[  412.185910] PC is at down_write+0x28/0x6c
[  412.185914] LR is at down_write+0x1c/0x6c
[  412.185917] pc : [<80878ee0>]    lr : [<80878ed4>]    psr: 60010013
[  412.185919] sp : b399bbc0  ip : b399bbc0  fp : b399bbd4
[  412.185921] r10: bc623660  r9 : 00000000  r8 : 80fee3b8
[  412.185924] r7 : 00000000  r6 : bc36e840  r5 : b995e438  r4 : 00000004
[  412.185926] r3 : ffff0001  r2 : 00000000  r1 : d1530275  r0 : 00000000
[  412.185930] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  412.185933] Control: 10c5383d  Table: 3b53006a  DAC: 00000055
[  412.185936] Process umount (pid: 1737, stack limit = 0xb399a210)
[  412.185939] Stack: (0xb399bbc0 to 0xb399c000)
[  412.185944] bbc0: 00000000 b995e438 b399bc84 b399bbd8 80488db0 80878ec4 00000000 00000000
[  412.185949] bbe0: 8148a6c0 00000001 b399bc84 b399bbf8 80234b40 801f87b0 bb530000 80488724
[  412.185954] bc00: 814054b0 00000001 81404d88 b399bc18 80235d74 80234a4c 00000000 81404dac
[  412.185958] bc20: ffffe000 00000001 80287344 00000000 00000001 00000000 be5dc4c4 01020200
[  412.185963] bc40: 00000000 20010013 00000000 00000000 8148ad40 d1530275 b399bca4 00000000
[  412.185968] bc60: bc36e840 81404d88 00000000 00000000 00000000 bc623660 b399bce4 b399bc88
[  412.185973] bc80: 804896b8 80488d5c bcdb9000 00000040 bcdb9000 00000000 b399bce4 b399bca8
[  412.185978] bca0: 80473320 80496c5c bccc2240 8026e008 b399bcdc b399bcc0 80292d28 d1530275
[  412.185982] bcc0: 00000000 bc36e840 bc623660 b995e438 bcdb9000 00000000 b399bd14 b399bce8
[  412.185987] bce0: 8048983c 80489678 00000000 bc623eb8 bc623eb8 0049a000 bc623eb0 004b0000
[  412.185992] bd00: bb868380 00000000 b399bd7c b399bd18 8026e0a4 804897f4 00000875 00000000
[  412.185997] bd20: bc36e840 00000000 00000000 b399bd38 00000800 8149dbe4 00000000 bc36e840
[  412.186002] bd40: 00000016 00000000 0049a000 7efea000 b399bd7c 00016000 00000005 bc36e840
[  412.186006] bd60: 0049a000 00000875 00000016 bb868380 b399bdbc b399bd80 8026e620 8026dd10
[  412.186011] bd80: 00000000 b399bddc bc3b6500 80878f40 bb8683b8 81404d88 bb8683b8 b399bddc
[  412.186016] bda0: bc36e840 0049a000 00016000 00000005 b399be14 b399bdc0 8025064c 8026e394
[  412.186021] bdc0: 00001812 00000000 00000000 b399bdd8 b399bddc 7edc9000 00000000 b399bddc
[  412.186026] bde0: b399bddc d1530275 7efff000 00005000 00016000 0049a000 00000fff 00000000
[  412.186030] be00: bba9e060 00000005 b399be2c b399be18 802506e4 802505b4 00001812 00000000
[  412.186035] be20: b399be54 b399be30 802f8064 802506a0 00001812 00000000 00000003 bbb8b600
[  412.186040] be40: 00000000 0049a000 b399befc b399be58 802f9894 802f7ff8 00001812 0001500c
[  412.186045] be60: 00000004 b9a10e70 00000000 ffffe000 fffff000 bbb8b634 00000000 bc3b6500
[  412.186050] be80: 00000000 00000000 00001812 00000000 ffffffff 81404d88 bba9e000 0049a000
[  412.186054] bea0: bc705f00 bc36e600 bc0bbc40 b399be58 00000034 00000000 802a1f34 b995e438
[  412.186059] bec0: 00000004 00000000 bb868380 d1530275 01111fc8 81427284 81517dc0 fffffff8
[  412.186064] bee0: 81424360 81517dc0 80a8f768 bc3b6500 b399bf34 b399bf00 802a35a0 802f941c
[  412.186069] bf00: b399bf34 00000002 802a2900 81404d88 bcfa4000 bbcacb00 bb868380 011145d0
[  412.186074] bf20: 000006c9 bc3b6500 b399bf84 b399bf38 802a40d8 802a3500 b399bf74 00000000
[  412.186079] bf40: 8149dbe4 bb8683b8 000006c9 00000000 00000000 d1530275 0000000b 011145d0
[  412.186083] bf60: 01111fc8 01114758 0000000b 801088a4 b399a000 00000000 b399bfa4 b399bf88
[  412.186088] bf80: 802a44d0 802a3b14 00000000 60010010 0058f3ec 01115338 00000000 b399bfa8
[  412.186093] bfa0: 80108700 802a449c 0058f3ec 01115338 01114758 011145d0 01111fc8 d58b7900
[  412.186098] bfc0: 0058f3ec 01115338 01114758 0000000b 00596d14 011145d0 01111fc8 76f66900
[  412.186103] bfe0: 0058f624 7ed10634 004cf668 76e4e4ac 60010010 01114758 00000000 00000000
[  412.186120] [<80878ee0>] (down_write) from [<80488db0>] (au_reval_and_lock_fdi+0x60/0x6f8)
[  412.186129] [<80488db0>] (au_reval_and_lock_fdi) from [<804896b8>] (au_write_pre+0x4c/0x12c)
[  412.186135] [<804896b8>] (au_write_pre) from [<8048983c>] (aufs_mmap+0x54/0x21c)
[  412.186144] [<8048983c>] (aufs_mmap) from [<8026e0a4>] (mmap_region+0x3a0/0x684)
[  412.186152] [<8026e0a4>] (mmap_region) from [<8026e620>] (do_mmap+0x298/0x45c)
[  412.186159] [<8026e620>] (do_mmap) from [<8025064c>] (vm_mmap_pgoff+0xa4/0xec)
[  412.186164] [<8025064c>] (vm_mmap_pgoff) from [<802506e4>] (vm_mmap+0x50/0x64)
[  412.186172] [<802506e4>] (vm_mmap) from [<802f8064>] (elf_map+0x78/0xc8)
[  412.186181] [<802f8064>] (elf_map) from [<802f9894>] (load_elf_binary+0x484/0x12c0)
[... cut ...]
[  412.231589] [<801090d0>] (arch_cpu_idle) from [<8087b69c>] (default_idle_call+0x34/0x48)
[  412.231596] [<8087b69c>] (default_idle_call) from [<80166db0>] (do_idle+0xf0/0x158)
[  412.231602] [<80166db0>] (do_idle) from [<801670d8>] (cpu_startup_entry+0x28/0x2c)
[  412.231608] [<801670d8>] (cpu_startup_entry) from [<8010eaac>] (secondary_start_kernel+0x160/0x16c)
[  412.231615] [<8010eaac>] (secondary_start_kernel) from [<001019ec>] (0x1019ec)
[  412.231619] ---[ end trace 8683963bba334e29 ]---

ZubairLK, 2019-01-10:
This is annoying. And quite tricky to debug (ideas welcome).
And now that https://github.com/balena-io/balena-supervisor/pull/866 is merged into the supervisor. It will trigger 'graceful' reboots. We'll see this more and more often.

And this can occasionally take 10+ minutes.
https://github.com/balena-os/meta-balena/pull/1349/commits/801c04122e087a498f636854c54c80e789ba6a8f can mitigate the impact. But is not a 'fix'.

We need to trace where there is a sleep/long timeout in the poweroff/reboot stage

ZubairLK, 2019-01-16:
https://github.com/balena-os/balena-raspberrypi/pull/276 updated the firmware and kernel for Pi 3A support.

I have been unable to reproduce this on the v2.29 dev image

ZubairLK, 2019-01-16:
Closing until we see this again on v2.29+

closed 2019-01-16 by ZubairLK