balena-os/balena-raspberrypi #291: Raspivid memory leak causes video to stop recording unexpectedly

bprovance, 2019-02-08:
I have been using the Raspberry Pi 3B+ as a camera to record video over the past few months, and I noticed that every now and then I would start a recording and within 30 seconds or so, my code would think that the camera module was recording but nothing was being written to disk. I was using `raspivid` to record and the process would not exit. I had to check the filesystem to see that the file size wasn't changing to determine this. I later watched the videos and confirmed that they were too short. 

This happened 4 times in about 200 recordings (typically 5-10 minute videos), but recently I noticed that if I let the Pi record video for 1 hour, I would hit the bug. I ran 8 tests and all 8 hit the bug in a range of 55-65 minutes. I was running BalenaOS 2.26.0 at the time. 

I ran `dmesg` and saw that each Pi had the message: `cma: cma_alloc: alloc failed, req-size: 1 pages, ret: -16` or something similar.

Realizing that this was likely a memory issue I tracked down the kernel release notes to discover that Linux kernel 4.14.68 (which BalenaOS 2.26.0 runs) contains a memory leak in the video port interface driver that would explain my issue. I verified this by recording with the same hardware on a RaspbianOS with Linux kernel 4.14.71 which contains a fix for the memory leak. On RaspbianOS I was able to record for 2+ hours with no issues. 

I spoke with Balena support and they kindly informed me that the latest BalenaOS release (2.29.2) runs Linux kernel 4.14.79. I upgraded two of my devices to this OS and repeated the test. I am happy to report that both Pis recorded video for over 4 hours without issue. 

My test was to run `raspivid -n -t 0 -rot 90 -w 720 -h 1280 -fps 24 -b 2000000 -o testVideo.h264`. I would guess that increasing the bitrate would cause one to hit the bug faster on the older kernel, but I have not tested this.

Linux kernel 4.14.71 release notes: https://lwn.net/Articles/765658/
```
* media: davinci: vpif_display: Mix memory leak on probe error path
* media: dw2102: Fix memleak on sequence of probes
```

bprovance, 2019-02-08:
Update: I ran the same raspivid function with the `-b 8000000` parameter and the bug was hit in under 20 minutes, as expected.

floion, 2019-02-08:
Hi @bprovance thanks for the heads-up. So this issue is fixed in the latest release which means this can be closed?

pdcastro, 2019-02-08:
@floion, the purpose of this issue is for us to patch earlier versions of balenaOS, so it stays open for that purpose. But you are right that it works with balenaOS 2.29.2. Thanks again @bprovance for reporting it!

bprovance, 2019-02-08:
No problem! I'm glad you guys updated the kernel :)

closed 2019-02-28 by floion