balena-os/balena-raspberrypi #300: Can't get touch screen to work

JimVanEeden, 2019-02-28:
I can not for the life of me get my Raspberry Pi 3 B+ with Raspberry Pi 7" Touchscreen to work. The display is working fine, the touch doesn't work at all. The hardware is not the problem, because it works fine when running on Raspbian. 

The app I'm running is an electron app, based on the electron-react template by you guys. Is this a common problem? If more information is needed I can provide it.

floion, 2019-02-28:
Hi. Does this touchscreen require any configuration in raspbian?

JimVanEeden, 2019-02-28:
No I'm actually testing it now and the *exact* thing runs straight away on Raspbian desktop.

floion, 2019-02-28:
Can you send a link to the exact touchscreen? Amazon link would be nice

JimVanEeden, 2019-02-28:
It's the official Raspberry Pi 7" Touchscreen:

https://www.kiwi-electronics.nl/7-inch-raspberry-pi-dsi-touchscreen-display?lang=en

floion, 2019-02-28:
Can you try this example we have? https://github.com/balena-io-projects/balena-dash

JimVanEeden, 2019-03-01:
Just did, no touch :(

floion, 2019-03-01:
@dansku can you chime in a bit since you have this working?

dansku, 2019-03-01:
@JimVanEeden I will setup the project from scratch tomorrow and give you feedback then.
In the meanwhile, could you post a picture of the driver board? I would like just to confirm an make sure it is the same as the one I have here. Cheers

floion, 2019-03-02:
Maybe it's a similar setup issue as https://github.com/balena-io-projects/balena-wpe/issues/38 ?

dansku, 2019-03-02:
@JimVanEeden I am running some tests, and this is not a final answer. Could you try setting up a WPE_URL address, this one, for example, https://www.balena.io/blog/page/2/ and see if it the touchscreen works. In case it doesn't work, try rebooting the device (you can do it from the dashboard). After it reboots, can you let me know if it works? On my current tests, after rebooting the touchscreen started working. I will dig more on this today!

JimVanEeden, 2019-03-04:
Thanks for your help guys, I'm going to try to get it working today!

floion, 2019-03-06:
@JimVanEeden any luck?

JimVanEeden, 2019-03-06:
Unfortunately I couldn't get it to work, so for now I'll just be using Raspbian Lite and might try again once our fleet gets too big to manage without your tools.

closed 2019-03-06 by JimVanEeden

marhoy, 2019-04-12:
I have this issue as well, what was the solution?
I just installed balenaos on a rpi3b+ with standard rpi touch screeen

floion, 2019-04-24:
Hi, what OS version did you try with?

marhoy, 2019-04-25:
@floion Thanks for following up!
Not sure what I did wrong, but I'm not able to reproduce the issue.
My rpi3b+ with touch screen is currently running the wpe-browser, and all is good :)

floion, 2019-04-25:
@marhoy that's great news
@JimVanEeden if you still want to give it a shot, please let us know. I now have this display with me so can try and debug your issues if any

bitmorse, 2019-05-09:
wpe-browser + official Rpi touch screen works here, but the Electron React example does not: display is fine, just no touch.

floion, 2019-05-09:
@bitmorse can you try and restart the container after it's been started for the first time?

bitmorse, 2019-05-09:
Here's a video of me restarting the container and showing all config: https://www.icloud.com/photos/#0nIDwE1-k21nzYVU_1lJM-29w

We use the stock balena-react-electron repo with changed react app code (just a few lines) in a multicontainer setup. Our docker-compose.yml:
```yaml
version: '2'
services:
#  balena_wpe:
#    privileged: true
#    build: ./balena-wpe
#    depends_on:
#      - user_app
  electron_frontend:
    build: ./electron-frontend
    network_mode: 'host'
    privileged: true
    ports:
      - 3000:3000
  node_backend:
    build: ./node-backend
    network_mode: 'host' #possibly required for slcan0 to work
    privileged: true
    #restart: always
    ports:
      - 80:80
      - 8080:8080
    devices:
      - "/dev/i2c-1:/dev/i2c-1"
      - "/dev/mem:/dev/mem"
      - "/dev/ttyACM0:/dev/ttyACM0"
    cap_add: 
      - SYS_RAWIO
    labels:
          io.balena.features.kernel-modules: '1'
          io.balena.features.firmware: '1'
          io.balena.features.dbus: '1'
          io.balena.features.supervisor-api: '1'
          io.balena.features.balena-api: '1'
          io.balena.update.strategy: download-then-kill
          io.balena.update.handover-timeout: '60000'
```

danielmahon, 2019-05-24:
try adding the env var `UDEV=1`