balena-os/balena-raspberrypi #304: error: no matching package named `atty` found (required by `cargo`)

siredmar, 2019-03-04:
In reference to #261 i reopen this issue because i have the same problem that i cannot fix.

siredmar, 2019-03-04:
To give some further information:

```
DEBUG: Executing shell function do_compile
+ cd /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git
+ do_compile
+ cargo_do_compile

+ cd /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/git
+ do_compile
+ cargo_do_compile
+ oe_cargo_fix_env
+ export CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cc
+ CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cc
+ export CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cxx
+ CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cxx
+ export 'CFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ CFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ export 'CXXFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ CXXFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ export AR=ar
+ AR=ar
+ export TARGET_CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cc
+ TARGET_CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cc
+ export TARGET_CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cxx
+ TARGET_CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cxx
+ export 'TARGET_CFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ TARGET_CFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ export 'TARGET_CFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ TARGET_CFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ export TARGET_AR=ar
+ TARGET_AR=ar
+ export HOST_CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/build-rust-cc
+ HOST_CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/build-rust-cc
+ export HOST_CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/build-rust-cxx
+ HOST_CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/build-rust-cxx
+ export 'HOST_CFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ HOST_CFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ export 'HOST_CXXFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ HOST_CXXFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ export HOST_AR=ar
+ HOST_AR=ar
+ oe_cargo_build
+ export 'RUSTFLAGS= -L /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/lib/rust -Zremap-path-prefix-from=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0 -Zremap-path-prefix-to=/usr/src/debug/cargo-native/0.25.0-r0'
+ RUSTFLAGS=' -L /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/lib/rust -Zremap-path-prefix-from=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0 -Zremap-path-prefix-to=/usr/src/debug/cargo-native/0.25.0-r0'
+ export RUST_TARGET_PATH=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/lib/rustlib
+ RUST_TARGET_PATH=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/lib/rustlib

+ oe_cargo_fix_env
+ export CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cc
+ CC=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cc
+ export CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cxx
+ CXX=/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/wrapper/target-rust-cxx
+ export 'CFLAGS=-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
+ CFLAGS='-isystem/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/recipe-sysroot-native/usr/include -O2 -pipe'
[... cut ...]
WARNING: /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/run.do_compile.924:1 exit 101 from '"/opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/cargo-0.25.0-x86_64-unknown-linux-gnu/cargo/bin/cargo" build -v --target x86_64-linux --release "$@"'
+ exit 101
ERROR: Function failed: do_compile (log file is located at /opt/projects/balena-raspberrypi/build/tmp/work/x86_64-linux/cargo-native/0.25.0-r0/temp/log.do_compile.924)

floion, 2019-03-11:
Can this be closed now since using 18.04 (a supported Poky OS) has fixed the issue for you?

siredmar, 2019-03-11:
It can be closed if you promise to adapt the documentation for balenaOs :)
Btw, it was Ubuntu **16.04** that solved the problem.

closed 2019-03-11 by siredmar