balena-os/balena-raspberrypi #306: Support for FT5426 displays

samuelcasas, 2019-03-06:
Hi, is there any workaround to get the touchscreen working? I thought the screen had an FT5406 chip but instead it has an FT5426.

floion, 2019-03-06:
Hi, what touchscreen is this? Can you send a link to see its specs?

samuelcasas, 2019-03-06:
It's the foundation 7" display, after few hours debugging it was recognized as FT5406 but the digitizer is a FT5426, and I confirmed the loading of the drivers by doing a `dmesg`:
```
dmesg | grep -i ft5406  
[    2.282902] rpi-ft5406 rpi_ft5406: Probing device  
[    2.286451] input: FT5406 memory based driver as /devices/virtual/input/input0
```

Also I did a `lsmod` and this is the output:
```
lsmod | grep -i ft  
rpi_ft5406              4783  0  
```

But the problem is that I want to rotate the screen and the touch, I did the rotation of the screen but the touch I didn't, I read about doing a rotate touch, I tried this in the `config.txt`:
```
display_lcd_rotate=2
dtoverlay=rpi-ft5406,touchscreen-inverted-x=1,touchscreen-inverted-y=1
```
But the device rebooted and erased those lines from `config.txt` and everything was like it was before I did those changes.

samuelcasas, 2019-03-06:
After doing some research the problem it might be is that I'm using X server as window manager as in [resin-electronjs](https://github.com/balena-io/resin-electronjs) and is not loading those changes correctly, I tried modifying the `start.sh` with `xinput set-prop 'FT5406 memory based driver' 'Evdev Axes Swap' 1` and it not worked.

floion, 2019-04-24:
@samuelcasas in order to rotate the display and the touch you would need to set display_lcd_rotate and the appropriate dtoverlay to handle the touch rotation also.
So in your balena-cloud dashboard you would go to the `Fleet configuration` page (if you set it from here all your devices will have this set) or the `Device Configuration` page for your device (this will add these changes only to 1 device) and add these 2 new variables:

`RESIN_HOST_CONFIG_display_lcd_rotate` with the value `1` for example
and
`RESIN_HOST_CONFIG_dtoverlay` with the value `rpi-ft5406,touchscreen-swapped-x-y=1,touchscreen-inverted-x=1` to match the display_lc_rotate example from above.

closed 2019-04-24 by floion