balena-os/balena-raspberrypi #310: Use kernel 4.9-skb for revpi-core-3

smilykoch, 2019-03-13:
Hi Balena.

We are using your software on a fleet of revpi-core-3, and had found a bug resulting in lots of recv timeouts. The bug was reported and fixed in the balena-revpi repo before it being archived. 

I have been promised that you will downgrade the kernel of the revpi to kunbus's 4.9-skb branch in this repo, but was wondering if you could give us a timeline on when we can expect this to happen, and also when we can expect to see a full released image including this change, for a balenaOS host update? We have a large number of costumers experiencing issues as a result of this, and would very much like to be able to hostupdate their devices remotely. This is thus a huge issue for us!

Thank you.

smilykoch, 2019-03-13:
@floion

floion, 2019-03-14:
Hi, looking to have this done and released sometimes next week.

smilykoch, 2019-03-17:
Sounds great! The sooner the better, as we have a lot of affected costumers.

smilykoch, 2019-03-20:
@floion Hi, any updates regarding this issue? Our costumers are pushing us here. Anything i can do to help?

floion, 2019-03-20:
Hi. We need some fixes in other parts of the OS done tomorrow and then we'll do the switch to the 4.9 kernel, test and hopefully deploy in the next days. Stay tuned, will  comment here when it's done

floion, 2019-03-21:
Created https://github.com/balena-os/balena-raspberrypi/pull/317

smilykoch, 2019-03-21:
Great to see a PR ! Remember though, that the piControl module is also built on a kernel specific branch.

smilykoch, 2019-03-24:
Any update on when we can expect a release of this, available through balena host update?

floion, 2019-03-24:
It has been released internally and we are testing it. Hopefully on Tuesday it will be available

martinjlowm, 2019-03-27:
Please make some progress! Our customers are complaining - Kind regards, Finn Hunneche, CEO Blackbird, Denmark

floion, 2019-03-27:
We are running the internal testsuite. We discovered a problem with the hostOS update so we need to have that fixed before releasing to production.

floion, 2019-03-29:
Version 2.31.5+rev2 has been released and contains kernel 4.9 with the realtime extension enabled

closed 2019-03-29 by floion