balena-os/balena-raspberrypi #313: standard_init_linux.go:195: exec user process caused "exec format error"

OdysLam, 2019-03-14:
Greetings,

I have been trying to configure Balena with EdgeX foundry services but with no luck so far.

Image: balenalib/raspberrypi3-64-ubuntu 

every command will produce "standard_init_linux.go:195: exec user process caused "exec format error"" 

Example:

Dockerfile

```
FROM balenalib/raspberrypi3-64-ubuntu
# Create a consul user and group first so the IDs get set the same way, even as
# the rest of this may change over time.

#RUN addgroup consul && \
#   adduser --system --ingroup consul consul

# standard mongo db data dir directories
RUN mkdir /data
RUN mkdir /data/db
RUN echo "this directory is reserved for EdgeX database files" > /data/db/README

# EdgeX shared directories
ENV EDGEX_BASE=/edgex
RUN mkdir $EDGEX_BASE
RUN mkdir $EDGEX_BASE/logs
RUN echo "this directory is reserved for EdgeX log files" > $EDGEX_BASE/logs/README

# Consul config and data directories
# The /consul/data dir is used by Consul to store state. The agent will be started
# with /consul/config as the configuration directory so you can add additional
# config files in that location.
RUN mkdir /consul
RUN mkdir /consul/config
RUN mkdir /consul/data
RUN echo "this directory is reserved for EdgeX Consul config files" > /consul/config/README
RUN echo "this directory is reserved for EdgeX data files" > /consul/data/README
RUN chown -R 100:1000 /consul

COPY static-services-config.json /consul/config

ENTRYPOINT /usr/bin/tail -f /dev/null

```
Compose: 

```
version:` '2.1'
volumes:
  db-data:
  log-data:
  consul-config:
  consul-data:
  portainer_data:

services:

  volume:
    build: .
    #   container_name: edgex-files
    networks:
      - edgex-network
    volumes:
      - db-data:/data/db
      - log-data:/edgex/logs
      - consul-config:/consul/config
      - consul-data:/consul/data
      
  consul:
    image: consul
    ports:
      - "8400:8400"
      - "8500:8500"
      - "8600:8600"
#   container_name: edgex-core-consul
    hostname: edgex-core-consul
    networks:
      edgex-network:
        aliases:
            - edgex-core-consul
    volumes:
      - db-data:/data/db
      - log-data:/edgex/logs
      - consul-config:/consul/config
      - consul-data:/consul/data
    depends_on:
      - volume  
```

ZubairLK, 2019-03-15:
@OdysLam It looks like you are trying to run `docker build .` on your x86 machine whereas the binaries in the container are a different architecture. You probably need to install qemu and binfmt in your host OS.
Here is a blog post with the info
https://www.balena.io/blog/building-arm-containers-on-any-x86-machine-even-dockerhub/

The balenalib repos already have qemu-aarch64-static so it is easy to just run them under qemu via binfmt.

```
zubairlk@zubair-xps-resin:~$ docker run --rm -i -t  balenalib/raspberrypi3-64-ubuntu /bin/bash
root@bbf45cc90e50:/# find /usr/ | grep qemu
/usr/bin/qemu-aarch64-static
root@bbf45cc90e50:/# exit
exit
zubairlk@zubair-xps-resin:~$ dpkg -l | grep binfmt
ii  binfmt-support                             2.1.6-1                                       amd64        Support for extra binary formats
zubairlk@zubair-xps-resin:~$ dpkg -l | grep qemu
ii  ipxe-qemu                                  1.0.0+git-20150424.a25a16d-1ubuntu1.2         all          PXE boot firmware - ROM images for qemu
ii  qemu                                       1:2.5+dfsg-5ubuntu10.34                       amd64        fast processor emulator
ii  qemu-block-extra:amd64                     1:2.5+dfsg-5ubuntu10.34                       amd64        extra block backend modules for qemu-system and qemu-utils
ii  qemu-efi                                   0~20160408.ffea0a2c-2                         all          UEFI firmware for virtual machines
ii  qemu-slof                                  20151103+dfsg-1ubuntu1.1                      all          Slimline Open Firmware -- QEMU PowerPC version
ii  qemu-system                                1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries
ii  qemu-system-arm                            1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (arm)
ii  qemu-system-common                         1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (common files)
ii  qemu-system-mips                           1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (mips)
ii  qemu-system-misc                           1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (miscelaneous)
ii  qemu-system-ppc                            1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (ppc)
ii  qemu-system-sparc                          1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (sparc)
ii  qemu-system-x86                            1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU full system emulation binaries (x86)
ii  qemu-user                                  1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU user mode emulation binaries
ii  qemu-user-static                           1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU user mode emulation binaries (static version)
ii  qemu-utils                                 1:2.5+dfsg-5ubuntu10.34                       amd64        QEMU utilities
zubairlk@zubair-xps-resin:~/resin/sample_apps/coral_usb$ 
```

OdysLam, 2019-03-15:
@ZubairLK 

Hey, thanks for the extensive reply, it makes sense. I thought that Balena automatically built the containers for the needed architecture, I guess that's when using the online process and not the local dev mode.

Also, what about the build functionality of balena cli? Is it possible to build them on the rpi itself (although it's highly non optimal). Will definately check it out.

One last question, since I will be using already built containers (Edgex), how can I know whether they will run on the Raspberry pi ? (I know that it's not balena relevant).

Anyhow, thanks!

ZubairLK, 2019-03-18:
@OdysLam it is possible to build the functionality on the device. Quite slow though.
From https://www.balena.io/docs/learn/develop/local-mode/ 
`It uses the Docker daemon on the device to build container images` 

If edgex provide containers for the armv6/7 architecture, they should run on the rpi.

I guess you can check by running balenaOS, ssh into the hostOS, then run
`balena run -i -t urlforedgexdocker /bin/sh`

If it gives an exec format error, the architecture is wrong.

OdysLam, 2019-03-18:
@ZubairLK  Actually I discovered that using the balena CLI it is possible to build locally on the machine and emulate it automatically (--emulate) and then forward it on the machine.
https://github.com/balena-io/balena-cli/blob/master/doc/cli.markdown#local-ssh-deviceip

I see that containers are working normally from the balena point of view. I now want to determine whether they actually work as intended. Will try to post a relevant guide or something in the future. 
Thank you so much

sgammon, 2019-03-31:
@OdysLam okay, I think I understand - the containers built and ran, but the service inside your container didn't start up? perhaps my issues have something to do with `alpine` then, which I am using in place of `ubuntu`.

I'll give your technique a try and see if switching to the same image helps. Thank you again