balena-os/balena-raspberrypi #339: Raspberry Pi PoE hat fan not working

msnelling, 2019-06-02:
Would it be possible to detect the use of the official Raspberry Pi PoE (power-over-ethernet) hat and enable the fan?
The Raspbian distro does this but I would prefer to use the BalenaOS obviously.

floion, 2019-06-10:
@msnelling we'll look into this
@vicgal did you get this hat?

vicgal, 2019-06-19:
@msnelling, could you please describe the use case in which you saw that the fan was not working?
It appears that the fan is thermally activated, and using the `balenaOS 2.36.0+rev2` I observed that it went on from time to time.
I also flashed the latest `Raspbian Stretch` and observed the same thing.

msnelling, 2019-06-19:
That's odd, I also see the fan come on in Raspbian but it wasn't in balenaOS. I'm not sure what version I was using but it was the current when I posted the issue. I'll have to have reflash and try again.

vicgal, 2019-06-19:
If you would like to test the fan functionality on the balenaOS, the following two sysfs entries are responsible for starting the fan:
`/sys/devices/platform/rpi-poe-fan@0/hwmon/hwmon0/pwm1`
`/sys/devices/platform/rpi-poe-fan@0/hwmon/hwmon0/def_pwm1'

They hold a PWM value representing the duty cycle values in a range from 0 to 255.
I don't know if it has any correlation to the thermal cooling states.

To activate the fan you could one of both:
`echo 200 > /sys/devices/platform/rpi-poe-fan@0/hwmon/hwmon0/pwm1`
or
`echo 200 > /sys/devices/platform/rpi-poe-fan@0/hwmon/hwmon0/def_pwm1'

msnelling, 2019-06-19:
Thanks, I'll take a look at this when I get another chance.

floion, 2019-06-25:
Closing until then. Feel free to open if you still see issues with it

monsterslo, 2021-09-07:
Hello, 
I'm having issues activating my poe+ hat fan with balena os.
I tried to activate it with the commands above but no such files exist.

HardwareHarry, 2021-09-11:
Agreed with @monsterslo; using a Raspberry 4 with Raspberry's own POE+ board.  Using supervisor 12.10.1 and balenaOS 2.83.10+rev1.

The POE+ hat doesn't ever seem to spin, even when the CPU temperature is 70+ degrees centigrade.  Under the Host OS and the containers, the directory /sys/devices/platform does not contain a "rpi-poe-fan@0" directory.  Given the original issue was 2+ years ago, has the device path changed or is it missing?  I am reluctant to let the Pi run 70-80 degrees constantly.  Many thanks!

root@#######:/sys/devices/platform# ls -l
total 0
drwxr-xr-x  4 root root    0 Jan  1  1970 3f1735a0.nvram
drwxr-xr-x  4 root root    0 Jan  1  1970 Fixed MDIO bus.0
drwxr-xr-x  3 root root    0 Jan  1  1970 arm-pmu
drwxr-xr-x  3 root root    0 Jan  1  1970 cpufreq-dt
drwxr-xr-x  4 root root    0 Jan  1  1970 emmc2bus
drwxr-xr-x  4 root root    0 Jan  1  1970 fixedregulator_3v3
drwxr-xr-x  4 root root    0 Jan  1  1970 fixedregulator_5v0
drwxr-xr-x  4 root root    0 Jan  1  1970 gpu
drwxr-xr-x  3 root root    0 Jan  1  1970 kgdboc
drwxr-xr-x  4 root root    0 Jan  1  1970 leds
drwxr-xr-x  3 root root    0 Jan  1  1970 phy
drwxr-xr-x  2 root root    0 Sep 11 15:03 power
drwxr-xr-x  4 root root    0 Jan  1  1970 reg-dummy
drwxr-xr-x  3 root root    0 Sep 11 13:55 regulatory.0
drwxr-xr-x 10 root root    0 Jan  1  1970 scb
drwxr-xr-x  4 root root    0 Jan  1  1970 sd_io_1v8_reg
drwxr-xr-x  4 root root    0 Jan  1  1970 sd_vcc_reg
drwxr-xr-x  4 root root    0 Jan  1  1970 serial8250
drwxr-xr-x  3 root root    0 Jan  1  1970 snd-soc-dummy
drwxr-xr-x 23 root root    0 Jan  1  1970 soc
drwxr-xr-x  3 root root    0 Jan  1  1970 timer
-rw-r--r--  1 root root 4096 Jan  1  1970 uevent
drwxr-xr-x  4 root root    0 Jan  1  1970 v3dbus

floion, 2021-12-14:
Closing this in favor of https://github.com/balena-os/balena-raspberrypi/issues/715

closed 2021-12-14 by floion