balena-os/balena-raspberrypi #341: Can't Boot RPI3 with USB Key after 2.29 (2.36 broken)

nunux13, 2019-06-06:
Hello,

If I download and flash 2.36 on a usbkey using Etcher, I can't Boot RPI3.
It work with 2.29.0, so I'm stuck at 2.29.0. (I tried some randoms between 2.29 and 2.36
and didn't find better.)

Regards,

Pierre

nunux13, 2019-06-07:
https://files.resin.io/resinos/raspberrypi3/2.36.0%2Brev2.prod/image/balena.img.zip
=> Flashed using BalenaEtcher to a SanDisk Ultra 16Go SDCARD: Boot OK to logo
=> Flashed using BalenaEtcher to a SanDisk Ultra USB 3.0: Boot KO to Uboot> shell

nunux13, 2019-06-07:
USB:0 5 USB Device found [...]
1 Storage Device found [...]
Found U-Boot script /boot.scr
Executing script [...]
Scanning MMC [...]
[Lots of error about MMC...]
ERROR: Could not find a resin inmage of any sort[...]
Loading resinOS_uEnv.dat from device partition 1
** Bad device[...]
** Bad device[...]
** Bad device[...]
Bad Linux ARM zImage magic!
SCRIPT FAIL: continuing ...
[more error about PXE]
Uboot>

nunux13, 2019-06-07:
https://resin-production-img-cloudformation.s3.amazonaws.com/resinos/raspberrypi3/2.31.0%2Brev0.prod/image/balena.img.zip
=> OK

https://files.resin.io/resinos/raspberrypi3/2.31.5%2Brev1.prod/image/balena.img.zip
=> OK

https://resin-production-img-cloudformation.s3.amazonaws.com/resinos/raspberrypi3/2.32.0%2Brev1.prod/image/balena.img.zip
=> KO

So the bug was introduced between 2.31.5 rev1 and 2.32.

nunux13, 2019-06-09:
Hello, I try to build from sources 2.31.5 rev 5 but the release is broken. It seem that 2.31.5 rev 2 to 2.31.5 rev 5 is broken and can't compile)

floion, 2019-06-11:
Hi, what is the compile error you get?

markhaehnel, 2019-06-30:
I have the exact same problems as @nunux13. Using 2.31.5 rev1 works but not the newer ones.

ZubairLK, 2019-07-02:
This is probably because we disabled usb scanning in u-boot to speed up the boot by ~2 seconds..
https://github.com/balena-os/balena-raspberrypi/pull/311
Which merged in 2.31.3 onwards.

We disabled usb scanning from meta-balena as well https://github.com/balena-os/meta-balena/pull/1432

I guess we can add a flag in `resinOS_uEnv.txt` and if someone wants usb scanning, they can enable it via there which will be checked via `env_resin.h`? @agherzan?

cc @floion

nunux13, 2019-07-02:
Many thanks for pointing me the direction. I'm going to try to rebuild last version with USB scan enable.

I keep you informed,

Best,

ZubairLK, 2019-07-02:
@nunux13 a better thing to do would be to simply revert that commit instead of going back to that old version.

Do you mind me asking how you use balenaOS and any background on using a usb vs sd card?

markhaehnel, 2019-07-02:
@ZubairLK Do you plan on reverting that commit and release it with the next version?

I personally use USB drives cause the random IOPS tend to be much higher and they're more reliable than the SD card. They break fast in development cycles with flashing often and writing a lot of data.

nunux13, 2019-07-02:
As @markhaehnel I seek for better reliability. Also, we plan to use a particular made USB switch, with selecting Key A or Key B for dual boot remote option and recovery.

ZubairLK, 2019-07-02:
@markhaehnel we don't plan on reverting the commit. It'll add 2 seconds to everyones boot time.
usb boot isn't something we test/support during releases. but there are people who find it useful in similar use cases as you describe.

I've made an issue to keep a track of this and see what we can do.
https://github.com/balena-os/meta-balena/issues/1575

nunux13, 2019-07-06:
Hey guy,

USB is back on last version doing this:

git clone https://github.com/balena-os/balena-raspberrypi.git
cd balena-raspberrypi/
git checkout tags/v2.38.0+rev1
git submodule update --init --recursive

vi layers/meta-balena/meta-balena-common/classes/resin-u-boot.bbclass

RESIN_UBOOT_DEVICE_TYPES ?= "**usb**"

./balena-yocto-scripts/build/barys -m raspberrypi3

Result is here, if you need it:
https://share.jaguar-network.com/f2742735-4744-4330-88e0-e7744bf5677c

Thanks you so much,

Best

nunux13, 2019-07-06:
The image i've cooked is working on USB and booting fast without MMC detection :-)

But I still need to manually extend the partition schema on USB device. Can you point where I can start hacking an auto resize script ?

Also, I need my image to support thoses wifi USB dongles:
0bda:818b RTL8192EU WIFI 2Ghz only with external antenna
0bda:c811 RTL8811CU 2Ghz and 5Ghz with external antenna

Can you point me some tutorials on how to do that ?
(I mean: How to add a USB driver to the yocto kernel)

Regards,

Pierre

floion, 2019-07-06:
Hi @nunux13 , yes that would be the way to enable USB boot. We are still discussing internally how to add this because we would not want to add it for everybody as it will increase the boot time for people not using usb boot.

About the wifi chipsets, I believe that currently the in-tree rtl driver from the kernel tree is pretty stable. Here's an example on how you could enable it: https://github.com/balena-os/balena-intel/blob/master/layers/meta-balena-genericx86/recipes-kernel/linux/linux-yocto_%25.bbappend#L144

About the second chipset, if there is no in-tree kernel driver you would need to add a recipe for the out of tree kernel module. An example would be this one: https://github.com/balena-os/balena-intel/tree/master/layers/meta-balena-genericx86/recipes-kernel/rtl8812au

damianpowell, 2019-07-29:
> Hey guy,
> USB is back on last version doing this:
> [snip]

I followed these instructions but after struggling with installing jq for a little while, I'm now hitting compilation error for contextify (incompatible function pointers, amongst others). Is there a plan to reintegrate this into the main balenaos with a `balena local configure` setting?

floion, 2019-07-29:
This will be fixed from our side and you can see the solution tracked here: https://github.com/balena-os/meta-balena/issues/1575
Please don't do other changes if you use balena-cloud as they will potentially be lost when doing hostOS updates

acostach, 2020-01-13:
@nunux13 @markhaehnel Can you please check with latest BalenaOS release 2.46.1+rev1 for the Pi3, it is available in dashboard for download. As you already know, host boot mode needs to be enabled in the OTP bits so that the usb-key can be scanned for the uboot binary: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md