balena-os/balena-raspberrypi #344: Add support for Raspberry Pi 4

zrzka, 2019-06-24:
https://www.raspberrypi.org/products/raspberry-pi-4-model-b/

Jaeyoung-Lim, 2019-07-12:
Hi, are there any ETAs or updates regarding support for the RPi4?

floion, 2019-07-12:
Hi. No ETA yet. It can be in the following week, we need this https://github.com/agherzan/meta-raspberrypi/pull/437 merged first

Jaeyoung-Lim, 2019-07-16:
@floion I can see that the https://github.com/agherzan/meta-raspberrypi/pull/437 ! I would be happy to test it out before the production release

agherzan, 2019-07-16:
@Jaeyoung-Lim There is some more work needed. We will update this thread when it's ready for a spin.

Jaeyoung-Lim, 2019-07-17:
@agherzan Awesome

MHarris021, 2019-07-29:
According to the status of https://github.com/agherzan/meta-raspberrypi/pull/437 it has now been merged. I there anything else required before balena OS can support running on an rpi4?

floion, 2019-07-30:
Hi, yes, there is an issue related to the production image not booting when `console=null` is supplied in the kernel cmdline. We are currently looking at this

MHarris021, 2019-07-30:
> Hi, yes, there is an issue related to the production image not booting when `console=null` is supplied in the kernel cmdline. We are currently looking at this

@floion thank you for the update

rjhuijsman, 2019-08-07:
Hi! Any updates in the last week? Any hint of an ETA?

(Sorry to be nagging, our project will get a big benefit from being able to upgrade to the Pi 4!)

floion, 2019-08-12:
No updates yet. This is still in the working queue

codefriar, 2019-08-16:
Looking forward to this! Thanks for your hard work Balena team!

agherzan, 2019-08-16:
If you feel like testing some early support you can build https://github.com/balena-os/balena-raspberrypi/tree/ag/rpi4 (be aware that it's a wip branch which might get rebased).

ollym, 2019-08-17:
@agherzan that branch hasn't been updated in 17 days. What's blocking the RPI4 branch being merged? Are there changes up-stream you're still waiting on?

ollym, 2019-08-20:
Ok, @agherzan do you have any indication of timeline? If we're talking 1-2 weeks then we'll wait. Anything more I'll build that branch and we'll use that for a project we need to deploy. Thanks

Jaeyoung-Lim, 2019-08-20:
@agherzan I share the same status with @ollym

gpulido, 2019-08-20:
@agherzan I'm also in the same status as @ollym  and @Jaeyoung-Lim

agherzan, 2019-08-20:
Hi. Good to see this interest. We are currently testing and finalising support. The optimist estimation for deploying is this week if testing reveals no major issues.

mario7754, 2019-08-22:
Hi @agherzan. Any news regarding the merge to master branch?

agherzan, 2019-08-22:
The support PR is up and in review.

mario7754, 2019-08-22:
Do you expect it to be merged today?

agherzan, 2019-08-22:
It might happen. The best way to confirm this is to follow the PR.

greaterfool, 2019-08-22:
Well. My timing could not have been more perfect, it seems.

mario7754, 2019-08-23:
Why isn't it merged? Is something still missing?

JeffJassky, 2019-08-23:
After merging can we expect it show up as a device option under the "Add a new device" section of the dashboard? Or will we still have to manually build?

floion, 2019-08-24:
Hi. Expect this to be available in the dashboard by Wednesday. I will try for Tuesday, depends on my workload

JeffJassky, 2019-08-25:
In the meantime does anybody have a raspberrypi4 image they can share? I'm running OSX and having trouble getting Yocto/oe to build. Ideally, an image that I can add a wpa_supplicant file to.

mario7754, 2019-08-28:
Any news @floion ?

floion, 2019-08-29:
Hi all, balenaOS version 2.41.0+rev4 is now available for RaspberryPi 4. Enjoy!

closed 2019-08-29 by floion

wlisac, 2019-08-29:
This is great - thank you! Looking forward to trying it out. 

Out of curiosity, are there any plans to support 32 bit as well like with Pi 3?

pwt, 2019-08-29:
@wlisac : you might want to take a look at the response to my question on the Balena forum:
  https://forums.balena.io/t/building-balenaos-for-rpi4/23993

Jaeyoung-Lim, 2019-08-29:
Works great!

mullcom, 2019-10-09:
Some help please. I get my pi4 only to boot loop when i test this out. I dont finding any information to solving it..

floion, 2019-10-09:
Hi, what exactly are you trying?