balena-os/balena-raspberrypi #349: How to get updates for other RPI3 based boards

adolfogc, 2019-06-30:
Yesterday, I noticed that an update to balenaOS v2.38.1+rev1 from 2.36.0+rev2 was available for the *Raspberry Pi 3* devices from balena.io, this update, however, wasn't available for other RPI3 based devices such as the *NPE X500 M3 (COMMUNITY)*. What is required for such boards to get these updates as well?

floion, 2019-06-30:
The rule for community boards such as this one is that when the baseline board gets updated, in this case rpi3, then the corresponding community board gets updated. So the board you mention will be updated too

adolfogc, 2019-06-30:
I tried to update one such device but the *"No compatible target versions available"* message appears instead.

floion, 2019-07-01:
Yes, as I mentioned above, it will be updated. I started now deploying this board to version 2.38.0+rev1 and it will be available in about 1 hour

closed 2019-07-01 by floion

adolfogc, 2019-07-01:
Thanks, @floion. I just updated one device to the latest version. In what order are the balenaOS updates rolled out for the RPi3 family?

floion, 2019-07-01:
You are welcome. Order is roughly: rpi3, rpi3 64 bits, rpi3 based community, revolution pi, rpi2, rpi1 and pi zero

adolfogc, 2019-09-28:
Hi @floion, I noticed that the NPE community board got stuck with an old version (no updates available). Do you know if newer versions will become available sometime soon? There's been two updates for RPi3 since the last time, but none for these other boards.

floion, 2019-09-30:
Hi. Sorry about that. It should have received the update when rpi3 was updated. Will do that now

adolfogc, 2019-10-03:
Hi @floion, thanks, they're available now.