balena-os/balena-raspberrypi #356: balenaFin WiFi driver hang

majorz, 2019-07-31:
This is something we see for the first time - a balenaFin device had its WiFi driver hanged and connectivity was lost. Persistent logging was enabled but more on the reason was lost because of log rotation. The only relevant entries are:
```
Jul 30 20:23:25 f76241e kernel: IOCTL is not allowed while the device is not present or hang
Jul 30 20:23:25 f76241e kernel: set mac address failed! status=-1, error_code=0x0
Jul 30 20:23:25 f76241e kernel: Set MAC address failed
```

floion, 2020-01-31:
@majorz since we did not get other reports of this issue and in the meantime we also updated the wifi driver, can we close this item?

majorz, 2020-02-05:
@floion closing this now

closed 2020-02-05 by majorz