balena-os/balena-raspberrypi #358: Remove rpi4 debug firmware version (start4cd.elf)

agherzan, 2019-08-06:
https://github.com/balena-os/balena-raspberrypi/blob/8ba12b8fe22f89e2d51301a9bb517b8e89964d15/layers/meta-balena-raspberrypi/recipes-bsp/bootfiles/bcm2835-bootfiles.bbappend#L12

imrehg, 2019-08-07:
Currently these files are deployed on e.g. a Pi3 as well, and not required:
```
-rwxr-xr-x 1 root root 2759172 Aug  7 09:40 start4.elf
-rwxr-xr-x 1 root root  762880 Aug  7 09:39 start4cd.elf
-rwxr-xr-x 1 root root 4716552 Aug  7 09:40 start4db.elf
-rwxr-xr-x 1 root root 3672712 Aug  7 09:40 start4x.elf
```

imrehg, 2019-08-09:
Also other files, @agherzan:
```
-rw-r--r-- 1 greg greg 3030 Aug  9 16:06 fixup4cd.dat
-rw-r--r-- 1 greg greg 6068 Aug  9 16:06 fixup4.dat
-rw-r--r-- 1 greg greg 9146 Aug  9 16:06 fixup4db.dat
-rw-r--r-- 1 greg greg 9148 Aug  9 16:06 fixup4x.dat
```
this is on RPi3 build with latest master with a meta-balena PR, just for reference.

floion, 2019-08-28:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/370
For the non-rpi4 boards having the rpi4 boot firmware present, we should open a different ticket

floion, 2019-08-28:
https://github.com/balena-os/balena-raspberrypi/issues/371

closed 2019-08-28 by floion