balena-os/balena-raspberrypi #375: Raspberry Pi 4 - HDMI Active, no output

codefriar, 2019-08-29:
Greetings!

First, thanks for your work!

I downloaded, etched and booted a rpi 4 4gb. It shows up in the dashbaord, is ssh'able etc. When the device boots, the attached monitor wakes up, but no ouput is displayed. It's clear the monitor thinks there will be /is data present, because of backlight bleedthrough. However, nothing but a black screen is seen.

I'm attempting to run the Balena Dash project on it. 

Please advise. Any other information that would be helpful, let me know and I'll get it.

floion, 2019-08-30:
Had tested on my side with https://github.com/balena-io-playground/x11-window-manager and that worked fine.
Have you done any changes to the config.txt through the dashboard?

codefriar, 2019-08-30:
No I haven’t

codefriar, 2019-08-30:
Update: I've gotten it to display the Balena logo / splash screen, but it never flips over to the X display to show the URL.

@floion What changes did you make / suggest to config.txt?

floion, 2019-08-31:
@codefriar what changes did you do to have the logo displayed? There should be nothing to do to have it display that, I'm just curious what was the problem in your case.
By showing the URL, you are referring to the balena dash project? I haven't tried that yet, will do that in the following days. I only tried the X11 window manager demo

codefriar, 2019-09-04:
@floion So in order to see the logo, I had to switch monitors. on my 4k i get nothing, on an older 1080p i see the logo. Neither gets to the point where it's displaying the dashbaord

codefriar, 2019-09-10:
@floion  just following up, any chance to try the balena dash project?

floion, 2019-09-11:
@codefriar there is an issue with balena dash currently, see https://github.com/balena-io-projects/balena-dash/issues/43

floion, 2019-10-04:
@codefriar did you get your rpi to appear in the dashboard? If not, could you use a development image and login over serial debug connection so we can check what's wrong there?

codefriar, 2019-10-04:
It does show in the dashboard.

I was having the issue where the boot graphic never shows on 4k, and on 1080p it loads the graphic, but never boots to the application (because the application needs gl drivers that were removed.)

willswire, 2019-10-25:
I also have been experiencing this same issue when running x11 applications on RPI4.  HDMI output is active (monitor stays alive) but no video content is output, and the application logs do not throw errors/warns etc.