balena-os/balena-raspberrypi #378: RPI4 /proc/cpuinfo missing hardware and model name

pdcastro, 2019-09-02:
As reported in this forum thread:
https://forums.balena.io/t/rpi4-proc-cpuinfo-missing-hardware-and-model-name/29080

pwt, 2019-09-02:
As commented in the forums: it’s different for 64-bit kernels. See `/sys/firmware/devicetree/base/serial-number` and `/sys/firmware/devicetree/base/model` instead.

dauledk, 2019-10-11:
My Balena experience is currently paused by this. Trying to use a Raspberry Pi 4, Node.js and `node-rpio`. [The related issue is this one](https://github.com/jperkin/node-rpio/issues/105). Are there any news - or ideas for a possible fix?

jeff-at-dwelo, 2019-12-02:
The problem actually runs deeper. `/proc/device-tree` is a symlink to `/sys/firmware/devicetree/base`, and the latter is not visible from within Docker containers. We need some way to readonly mount that host path into the container.

hraftery, 2022-03-24:
> We need some way to readonly mount that host path into the container

Indeed. This directory is pretty important for 64-bit. I think more and more software will (and should) rely on being able to read it.

I wasn't able to find any combination of mount, bind, volume, etc. that worked. But `--privileged` automatically mounts it. Pity to have to use it, but is the only way I've found at the moment.