balena-os/balena-raspberrypi #379: RPi 3 audio and hdmi rotation possibly affected by 2.41.0 RPi 4 updates

pdcastro, 2019-09-03:
From support interactions:
> This is something from the kernel update apparently: https://github.com/raspberrypi/linux/issues/3181
> If you use the balena web dashboard, go to “Device Configuration” section, then modify the existing RESIN_HOST_CONFIG_dtoverlay variable to “vc4-fkms-v3d” then audio will be working again.
> It seems we need to also update the RPi firmware.

floion, 2019-10-04:
@pdcastro this should not be the case anymore. Please have a test of the latest 2.43 release and close this if the issue is fixed

pdcastro, 2019-10-07:
A forum user has confirmed that balenaOS 2.43 fixes the issue:
https://forums.balena.io/t/raspberry-pi-headphone-jack-no-longer-works-with-latest-os/27667/27?u=pdcastro

closed 2019-10-07 by pdcastro