balena-os/balena-raspberrypi #380: u-boot can't write bootcount.env

ZubairLK, 2019-09-04:
Probably since we moved to bsp u-boot which bumped the version.
u-boot is unable to use fatwrite to make a new file. The error message is `Error: allocating new dir entry`

It can still write if a file exists.. Log/reproduction steps below

```
U-Boot> env set bootcount 1
U-Boot> env export -t $kernel_addr_r bootcount
U-Boot> md $kernel_addr_r 20
00080000: 746f6f62 6e756f63 0a313d74 0f030f00    bootcount=1.....
00080010: 0f030f03 0f030f03 0f030f03 0f030f03    ................
00080020: 0f030f03 0f030f03 01030f03 03000150    ............P...
00080030: 03030303 03030303 03030303 03030303    ................
00080040: 03030303 03030303 03030303 41030303    ...............A
00080050: 03030303 03030303 03030303 03030303    ................
00080060: 03030303 03030303 03030303 03030303    ................
00080070: 03030303 03030303 03030303 03030303    ................
U-Boot> fatls mmc 0:1
    23315   bcm2708-rpi-0-w.dtb
    24205   bcm2708-rpi-b-plus.dtb
    23946   bcm2708-rpi-b.dtb
    23723   bcm2708-rpi-cm.dtb
    25293   bcm2709-rpi-2-b.dtb
    27082   bcm2710-rpi-3-b-plus.dtb
    26463   bcm2710-rpi-3-b.dtb
    25261   bcm2710-rpi-cm3.dtb
        0   bcm2835-bootfiles-20181112.stamp
      441   boot.scr
    52296   bootcode.bin
      260   config.json
    36254   config.txt
     2422   device-type.json
     6729   fixup.dat
     2644   fixup_cd.dat
     9803   fixup_x.dat
       40   image-version-info
   442340   kernel7.img
      204   os-release
            overlays/
       24   resin-image
     8677   resinos.fingerprint
            splash/
  2878340   start.elf
   685572   start_cd.elf
  3792552   start_x.elf
            system-connections/
       38   resinOS_uEnv.txt
   762880   start4cd.elf
     6068   fixup4.dat
    24407   bcm2708-rpi-zero-w.dtb
  4716552   start4db.elf
     9146   fixup4db.dat
  2759172   start4.elf
        0   bcm2835-bootfiles-20190709.stamp
    40559   bcm2711-rpi-4-b.dtb
     3030   fixup4cd.dat
  3672712   start4x.elf
     9148   fixup4x.dat
       86   cmdline.txt
        0   bcm2835-bootfiles-20190828.stamp
        0   bcm2835-bootfiles-20190718.stamp

40 file(s), 3 dir(s)

U-Boot> fatwrite mmc 0:1 $kernel_addr_r bootcount.env 0xd
Error: allocating new dir entry

** Unable to write "bootcount.env" from mmc 0:1 **
U-Boot> fatwrite mmc 0:1 $kernel_addr_r resin-image 0xd
13 bytes written
U-Boot> fatload mmc 0:1 $kernel_addr_r resin-image
13 bytes read in 2 ms (5.9 KiB/s)
U-Boot> md $kernel_addr_r
00080000: 746f6f62 6e756f63 0a313d74 0f030f00    bootcount=1.....
00080010: 0f030f03 0f030f03 0f030f03 0f030f03    ................
00080020: 0f030f03 0f030f03 01030f03 03000150    ............P...
00080030: 03030303 03030303 03030303 03030303    ................
00080040: 03030303 03030303 03030303 41030303    ...............A
00080050: 03030303 03030303 03030303 03030303    ................
00080060: 03030303 03030303 03030303 03030303    ................
00080070: 03030303 03030303 03030303 03030303    ................
U-Boot> 
```

ZubairLK, 2019-09-04:
```
U-Boot> version
U-Boot 2019.01-dirty (Sep 04 2019 - 16:25:13 +0000)

arm-poky-linux-gnueabi-gcc (GCC) 8.3.0
GNU ld (GNU Binutils) 2.32.0.20190204
U-Boot> 
```

acostach, 2020-12-18:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/471 which has been present in production for some time now. Issue no longer occurred with this fix.

closed 2020-12-18 by acostach