balena-os/balena-raspberrypi #383: Update VLI firmware on host OS update

floion, 2019-09-09:
The rpi foundation released a VLI firmware update in order to mitigate several issues, one of them being overheating: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=250990

We need to decide if we include this early testing version

acostach, 2019-10-10:
@floion Currently waiting for a few more days to get an answer from rpi engineers on when and if this early testing version will become a beta or stable release. The PR to add support for this is ready, just needs the final firmware binary source to be specified.

floion, 2019-10-10:
Sounds good, thank you

acostach, 2019-10-14:
Answer from maintainers:

"_Any supported updates to the VLI will be via the **rpi-eeprom** package we haven't decided on the update mechanism yet which won't necessarily be the vl805 tool.
Sorry no timescales yet, maybe a beta in the next few weeks_"

Discussed in upstream issue: https://github.com/raspberrypi/firmware/issues/1270

Will also update PR to re-use rpi-eeprom package I added for bootloader code update, which is merged. Will do so as soon as the upstream rpi-eeprom repository gets the beta firmware.