balena-os/balena-raspberrypi #384: Support bootloader firmware update

floion, 2019-09-09:
The rpi4 does not use the bootcode.bin in the boot partition anymore: https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

Instead, raspbian seems to be using a package called rpi-eeprom for updating the bootcode which now lives in a small spi eeprom. We should probably add this package too as part of the host OS update procedure.

acostach, 2019-10-25:
In the case of hup to a release that contains a kernel version change, modprobe will fail when executed from the container as it looks for the module in the path of the current kernel version. This will trigger a rollback and ends in failed hup.

Not including this change currently in order to adapt for this use-case too.

philkunz, 2020-04-29:
Is there any update on this issue?

floion, 2020-05-01:
Hi @philkunz not yet, we need to do some refactoring in other areas before we add this.
See https://github.com/balena-os/balena-raspberrypi/commit/16ac8ab30bdb698d2f886eb6796ce413df3bd74e for an explanation why we removed it

floion, 2020-05-01:
cc @alexgg @mtoman ^