balena-os/balena-raspberrypi #386: balenaCloud Fin instructions different from documentation

dansku, 2019-09-09:
https://github.com/balena-os/balena-raspberrypi/blob/1f1064e1272418314549b7df06a07a67efe94d0a/fincm3.coffee
[image]

In BalenaCloud setup, it says to power the BalenaFin with barrel jack AND plug in USB.
In the BalenaFin documentation, it says to ONLY power the fin from USB when flashing. 

Fin documentation and dashboard instructions disagree on how to power the fin during flashing, balena.io/fin/1.1/docs says to only use the prg port (1) while the balenaCloud instructions mention both, see user-provided screenshots in this forum thread (2)

(1): https://github.com/balena-io/resin-site/blob/master/src/data/fin/1.1/docs/getting-started.md#flashing-the-os
(2): https://forums.balena.io/t/31491

ntzovanis, 2019-09-09:
This is correct, for flashing the v1.0 Fin, both Jack and USB have to be connected. In case of the v1.1 Fin, only USB.

floion, 2020-01-31:
@garethtdavies ^ see above for some nuance in the docs.

floion, 2020-03-05:
Fixed by https://github.com/balena-os/balena-raspberrypi/pull/466

closed 2020-03-05 by floion