balena-os/balena-raspberrypi #393: Release updated build for Pi v1

davide125, 2019-09-15:
BalenaOS for Raspberry Pi v1 / Zero is currently at 2.32.0+rev1, while other boards are at 2.38.0+rev1 or  2.41.0+rev4. Would it be possible to release an updated build for the Pi v1 as well? Specifically, I'm hitting https://github.com/balena-io/balena-supervisor/issues/970 which has been fixed in v2.33.0 and later (via https://github.com/balena-io/balena-supervisor/pull/964). I also hit https://github.com/balena-io/balena-cli/issues/1355 but I'm not positive that's OS related (and it only affects the local workflow anyway). Thanks!

floion, 2019-09-16:
Hi, yes, this board will receive the new OS by end of next week I estimate.

floion, 2019-10-23:
This now has OS version 2.44.0+rev1

closed 2019-10-23 by floion

huckym, 2019-12-06:
Has it again reverted back to 2.32.0+rev1? I can't seem to find any discussion around this. May I know the reason?

floion, 2019-12-09:
Hi. There was a bug in that release which made the supervisor not start sometimes. That is the reason the release got pulled off

huckym, 2019-12-17:
> Hi. There was a bug in that release which made the supervisor not start sometimes. That is the reason the release got pulled off

got it. thanks for the reply