balena-os/balena-raspberrypi #395: Add RaspPi Zero W to "Device Type" Dropdown List

dtischler, 2019-09-18:
Original Issue was logged here:  https://github.com/balena-io/resin-ui/issues/3067, 

After discussion with @thgreasi , moving to this Repo.

**Expected Behavior**
It would be nice to add the Raspberry Pi Zero W to the list of Device Types that is presented when creating a new Application. We currently list the Raspberry Pi v1 and Zero, but customer get confused and then assume we do not support the Zero W because it is not explicitly listed.

**Actual Behavior**
As described

**Steps to Reproduce the Problem**
Create Application
Dropdown menu lists the possible Device Types to choose from
Specifications
N/A

Browser:
OS:
Environment:

[image]

thgreasi, 2019-09-18:
Essentially this is just about changing
https://github.com/balena-os/balena-raspberrypi/blob/078b0add0345e94f27c15d98256f8c7b61b66761/raspberrypi.coffee#L8
to something like `Raspberry Pi (v1 / Zero / Zero W)` (not sure about the wording).
Let me know if this is harder that I think or has other implications (did we have any when the Pi Zero was added?).

floion, 2019-10-04:
@shaunmulligan how shall the wording for this one be?

shaunmulligan, 2019-10-10:
@floion the wording @thgreasi suggests looks fine to me for now

closed 2019-10-18 by floion