balena-os/balena-raspberrypi #399: Raspberry Pi 4 and raspistill breaking with mmal issues

imrehg, 2019-09-24:
Tested 2.41.0+rev4 Raspberry Pi4 with raspistill, it is not snapping pictures, and does these errors:

```
24.09.19 11:50:16 (+0100)  camera-new  snapping a picture via raspistill -o /data/test_new_20190924_105016.jpg
24.09.19 11:50:16 (+0100)  camera-new  mmal: mmal_vc_shm_init: could not initialize vc shared memory service
24.09.19 11:50:16 (+0100)  camera-new  mmal: mmal_vc_component_create: failed to initialise shm for 'vc.camera_info' (7:EIO)
24.09.19 11:50:16 (+0100)  camera-new  mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7)
24.09.19 11:50:16 (+0100)  camera-new  mmal: Failed to create camera_info component
```

[Flowdock conversation](https://www.flowdock.com/app/rulemotion/resin-devices/threads/-X7G6a8-qmZgsgqSAHXNIe5lb-b)

imrehg, 2019-09-25:
Actually, seems like it still takes a picture properly, so this is more of a warning? But unsure where is it coming from, or how to fix it yet.

DavideFemia, 2021-05-04:
i had similar problem with balenaOS 2.73.1+rev1 on rpi4 when i execute camera = PiCamera() from python:

mmal: mmal_vc_shm_init: could not initialize vc shared memory service
mmal: mmal_vc_component_create: failed to initialise shm for 'vc.camera_info' (7:EIO)
mmal: mmal_component_create_core: could not create component 'vc.camera_info' (7)
Traceback (most recent call last):
File "src/main.py", line 21, in <module>
camera = PiCamera()
File "/usr/local/lib/python3.6/site-packages/picamera/camera.py", line 367, in __init__
with mo.MMALCameraInfo() as camera_info:
File "/usr/local/lib/python3.6/site-packages/picamera/mmalobj.py", line 2346, in __init__
super(MMALCameraInfo, self).__init__()
File "/usr/local/lib/python3.6/site-packages/picamera/mmalobj.py", line 633, in __init__
prefix="Failed to create MMAL component %s" % self.component_type)
File "/usr/local/lib/python3.6/site-packages/picamera/exc.py", line 184, in mmal_check
raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.camera_info': I/O error

the rpi4 was working properly for 5 days straight when suddenly it stopped after pushing new code without updating Dockerfile.template.