balena-os/balena-raspberrypi #401: Move resin-data to external drive

bbqsrc, 2019-09-27:
Is this possible? I can find no documentation about it either way.

I have concerns about an SD card hosting entire Docker environments and have an M.2 enclosure just sitting there begging to be used. :smile:

floion, 2019-09-27:
Hi. So just to clarify, you do not want to have an additional storage drive as described here https://forums.balena.io/t/best-practise-to-use-an-external-drive/21739 but rather replace the data partition from the sd card to be on external drive?

bbqsrc, 2019-09-27:
Yeah. My logic is that it would simplify the updating process, as then I can just change SD cards to update the balena OS while my data remains persistent on a separate disk.

floion, 2019-09-27:
Unfortunately that is not something we support at the moment. Will try and see if we can get that to some sort of brainstorm

bbqsrc, 2019-09-27:
Cool. How does the current partition layout get generated? /etc/fstab is very empty. :smile:

floion, 2019-09-27:
It's done at build time by this class https://github.com/balena-os/meta-balena/blob/development/meta-balena-common/classes/image_types_resin.bbclass

bbqsrc, 2019-09-27:
Okay, how does the OS mount them? Blindly mounting by partition label?

floion, 2019-09-30:
Here are the yocto recipes for doing that: https://github.com/balena-os/meta-balena/tree/development/meta-balena-common/recipes-support/resin-mounts

And the relevant systemd units are here https://github.com/balena-os/meta-balena/tree/development/meta-balena-common/recipes-support/resin-mounts/resin-mounts

agherzan, 2019-10-02:
@bbqsrc you will probably find this helpful: https://github.com/balena-os/meta-balena/pull/1677

nonnos, 2021-02-24:
No news regarding this? @bbqsrc what was your final solution?

bbqsrc, 2021-02-25:
Ended up not being able to use balena at all. 🙂