balena-os/balena-raspberrypi #417: [pi4] wifi-connect doesn't work. kernel errors as well

ZubairLK, 2019-10-31:
Some journalctl logs. 

My Android phone keeps trying to connect/disconnect
```
Oct 31 12:49:01 balena wpa_supplicant[898]: wlan0: AP-STA-DISCONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:02 balena wpa_supplicant[898]: wlan0: AP-STA-CONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:02 balena wpa_supplicant[898]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Oct 31 12:49:05 balena wpa_supplicant[898]: wlan0: AP-STA-DISCONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:06 balena wpa_supplicant[898]: wlan0: AP-STA-CONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:06 balena wpa_supplicant[898]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Oct 31 12:49:09 balena wpa_supplicant[898]: wlan0: AP-STA-DISCONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:10 balena wpa_supplicant[898]: wlan0: AP-STA-CONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:10 balena wpa_supplicant[898]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Oct 31 12:49:12 balena wpa_supplicant[898]: wlan0: AP-STA-DISCONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:13 balena wpa_supplicant[898]: wlan0: AP-STA-CONNECTED 0e:78:2c:fe:34:32
Oct 31 12:49:13 balena wpa_supplicant[898]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Oct 31 12:49:16 balena wpa_supplicant[898]: wlan0: AP-STA-DISCONNECTED 0e:78:2c:fe:34:32
Oct 31 12:50:03 balena wpa_supplicant[898]: wlan0: AP-STA-CONNECTED 0e:78:2c:fe:34:32
Oct 31 12:50:03 balena wpa_supplicant[898]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Oct 31 12:50:06 balena wpa_supplicant[898]: wlan0: AP-STA-DISCONNECTED 0e:78:2c:fe:34:32
```

The message  `brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52` is from kernel
```
root@balena:~# dmesg | tail
[   17.496821] brcmfmac: power management disabled
[   17.626911] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   35.162879] brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[   35.169957] brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[   35.183635] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   35.763765] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   35.770050] brcmfmac: power management disabled
[   36.339784] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   36.345864] brcmfmac: power management disabled
[   36.468009] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

exact versions of things
```
root@balena:~# uname -a
Linux balena 4.19.71 #1 SMP PREEMPT Fri Oct 25 14:28:11 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
root@balena:~# nmcli -v
nmcli tool, version 1.20.2
root@balena:~# 
root@balena:~# wpa_supplicant -v
wpa_supplicant v2.7
Copyright (c) 2003-2018, Jouni Malinen <j@w1.fi> and contributors
root@balena:~# cat /etc/os-release 
ID="balena-os"
NAME="balenaOS"
VERSION="2.44.0+rev3"
VERSION_ID="2.44.0+rev3"
PRETTY_NAME="balenaOS 2.44.0+rev3"
MACHINE="raspberrypi4-64"
VARIANT="Development"
VARIANT_ID="dev"
META_BALENA_VERSION="2.44.0"
RESIN_BOARD_REV="a4f4d1a"
META_RESIN_REV="7fed82f"
SLUG="raspberrypi4-64"
root@balena:~# 
```

same behaviour on a v2.45 build with wpa_supplicant v2.9

ZubairLK, 2019-10-31:
this is on a 4gb pi4

ZubairLK, 2019-11-04:
Saw something on rpi3 as well.

```
root@balena:~# dmesg | grep vndr
[ 4679.439380] brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[ 4679.446431] brcmfmac: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
root@balena:~# cat /etc/os-release 
ID="balena-os"
NAME="balenaOS"
VERSION="2.44.0+rev4"
VERSION_ID="2.44.0+rev4"
PRETTY_NAME="balenaOS 2.44.0+rev4"
MACHINE="raspberrypi3"
VARIANT="Development"
VARIANT_ID="dev"
META_BALENA_VERSION="2.44.0"
RESIN_BOARD_REV="aa51ce9"
META_RESIN_REV="8341c71"
SLUG="raspberrypi3"
root@balena:~# wpa_supplicant -v
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors
root@balena:~# nmcli --version
nmcli tool, version 1.20.2
uname -a
Linux balena 4.19.71 #1 SMP Wed Oct 30 13:20:39 UTC 2019 armv7l armv7l armv7l GNU/Linux
root@balena:~# 
```

vicgal, 2020-01-06:
Wasn't able to reproduce this on a PI4 but it seems that this issue has happened before on a PI3, https://github.com/volumio/Volumio2/issues/1462, https://github.com/raspberrypi/linux/issues/1342