balena-os/balena-raspberrypi #418: pi4 64bit os not connecting to 5g wifi network (2.4 okay)

gidjituser, 2019-11-04:
cannot connect to multiple 5g networks on multiple PI4's. Using latest os. 2.4 seems to work find. 

dmesg
[   19.192358] brcmfmac: power management disabled
[   44.008818] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   44.557136] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

nmcli info
GENERAL.DEVICE:                         wlan0
GENERAL.TYPE:                           wifi
GENERAL.HWADDR:                         F6:55:AA:AA:AA:AA
GENERAL.MTU:                            1500
GENERAL.STATE:                          30 (disconnected)
GENERAL.CONNECTION:                     --
GENERAL.CON-PATH:                       --

nmcli con show
Wired connection 1  8f677aa3-eedb-39ed-97d5-b11eafcc443c  ethernet  eth0
supervisor0         cd69f527-996b-48ca-a728-8e23d4b3024a  bridge    supervisor0
myasus5g            6669052a-13ae-44a3-a55d-abfb577bee02  wifi      --

system configuration 
root@a46aff9:~# uname -a
Linux a46aff9 4.19.71 #1 SMP PREEMPT Fri Oct 25 14:00:45 UTC 2019 aarch64 aarch64 aarch64 GNU/Linux
root@a46aff9:~# nmcli -v
nmcli tool, version 1.20.2
root@a46aff9:~# wpa_supplicant -v
wpa_supplicant v2.7
Copyright (c) 2003-2018, Jouni Malinen <j@w1.fi> and contributors
root@a46aff9:~# cat /etc/os-release 
ID="balena-os"
NAME="balenaOS"
VERSION="2.44.0+rev3"
VERSION_ID="2.44.0+rev3"
PRETTY_NAME="balenaOS 2.44.0+rev3"
MACHINE="raspberrypi4-64"
VARIANT="Production"
VARIANT_ID="prod"
META_BALENA_VERSION="2.44.0"
RESIN_BOARD_REV="a4f4d1a"
META_RESIN_REV="7fed82f"
SLUG="raspberrypi4-64"

floion, 2019-11-04:
Can you also try with the new firmware? https://github.com/RPi-Distro/firmware-nonfree/issues/5
We have not yet included it but we will

floion, 2020-01-31:
Hi, can you try to reproduce this with the latest image available? This latest release has the updated wifi firmware

Poltergeisen, 2020-08-29:
Mine isn't displaying when using the newest development image I downloaded today. Was there something I needed to enable? Or a certain wifi configuration needed from my router?

acostach, 2020-08-31:
Hi @Poltergeisen 

I checked now with the 2.54.3+rev3 Pi4 image from staging and I'm able to connect to a 5GHz ap:

```
root@f60831f:~# iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr dc:a6:32:c2:20:fb
                ssid Balena5GHz
                type managed
                channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
                txpower 31.00 dBm
```

This image has an updated firmware as well as the 5.4 kernel, and the boot firmware on my Pi is the stable one from 31 July - pieeprom-2020-07-31.bin.

Does the 2.54.3+rev3 image work for you?