balena-os/balena-raspberrypi #419: WiFi country setting does not work

majorz, 2019-11-04:
The `country` config.json setting does not work for RPi and Fin devices.

Internally it is set with `iw reg set <COUNTRY>` and it could be verified that it does not work with `iw reg get`.

I tracked this down to broken installation of the `crda` and `wireless-regdb` packages.

First, the `regulatory.db` file is missing. The following kernel message indicates that: `cfg80211: failed to load regulatory.db`. It needs to be installed at `/lib/firmware/regulatory.db` together with its signature file - `regulatory.db.p7s`.

Additionally there is a `85-regulatory.rules` file which is dislocated. Instead of being in `/lib/udev/rules.d/` with the rest of the udev rules, it is sitting alone in `/usr/lib/udev/rules.d/`.

On newer systems (using kernel 4.15+), the `crda` udev driver userspace package is not necessary anymore as the kernel itself loads the `regulatory.db` file.

I tested this by downloading `regulatory.db` and `regulatory.db.p7s` from the `wireless-regdb` repository and placing those in `/lib/firmware`. The `country` setting starts working afterwards.

Also tested the old way with `crda` by moving `85-regulatory.rules` to `/lib/udev/rules.d/` -- the `country` settings works in that way as well.

If we would like to keep using `crda` because of older kernel compatibility we need to fix the location of the udev rule. For newer kernels the `regulatory.db*` files need to be present in the `/lib/firmware` folder.

I tested this on Raspbian as well, where things are configured correctly.

References:
https://www.kernel.org/doc/Documentation/networking/regulatory.txt
https://github.com/agherzan/meta-raspberrypi/commit/414831c102849246fecab0f0cf7dcb9a3494d3fc

tommi-breezi, 2019-11-05:
majorz et al. - What would be the best way for me to get country setting working for my limited scale international tests? I don't mind if I need to do manual configuration for each of device for now.
Thanks!

majorz, 2019-11-06:
@tommi-breezi you may do that for testing purposes indeed. Please note that when the OS gets updated those changes will be lost. Since most of the host OS filesystem is in read-only mode you will need to remount it as read-write before applying the changes, e.g. `mount -o remount, rw /`.

tommi-breezi, 2019-11-07:
> @tommi-breezi you may do that for testing purposes indeed. Please note that when the OS gets updated those changes will be lost. Since most of the host OS filesystem is in read-only mode you will need to remount it as read-write before applying the changes, e.g. `mount -o remount, rw /`.

majorz - could you list the steps you did to get the wifi country setting to work?
-thanks

majorz, 2019-11-07:
@tommi-breezi 
The easiest one would be:
1. `mount -o remount, rw /`
2. `mv /usr/lib/udev/rules.d/85-regulatory.rules /lib/udev/rules.d/`
3. `reboot now`

Please let me know if that does not work for you.

floion, 2020-02-29:
Fixed by https://github.com/balena-os/meta-balena/pull/1843

AlidaOdendaal, 2020-03-02:
The PR (https://github.com/balena-os/meta-balena/pull/1843) solving this issue has not been merged yet, so re-opening the issue.

floion, 2020-03-27:
This has now been deployed to production for rpi4 and the rest will follow. The boards running OS version 2.48.0+rev1 or later will have this fixed. Closing this issue here

closed 2020-03-27 by floion